project-chip/connectedhomeip #23338: Add a way to do ExtractVIDPIDFromX509Cert via the Darwin framework

bzbarsky-apple, 2022-10-25:
This is needed to figure out the VID/PID of the device, as claimed in its DAC, after device attestation completes.

bzbarsky-apple, 2022-10-25:
Perhaps this API shape:
```
typedef NSData MTRDeviceAttestationCertificate;

@interface MTRProductIdentity : NSObject / * Name TBD */
@property (nonatomic, readonly) NSNumber * vendorID;
@property (nonatomic, readonly) NSNumber * productID;

- (instancetype)productIdentityFromDeviceAttestationCertificate:(MTRDeviceAttestationCertificate *)deviceAttestationCertificate error:(NSError * __autoreleasing *)error;

- (instancetype)init NS_UNAVAILABLE;
+ (instancetype)new NS_UNAVAILABLE;
@end
```

or so.  Or should the "From" there be "With"?

woody-apple, 2022-12-08:
I’d vote with. :)

PR #24335 (2023-01-10): Darwin: Add vendorID and productID properties to MTRDeviceAttestationDeviceInfo
Fixes #23338

Also add Xcode header templates for new files (due to an Xcode bug an extra `//` ends up at the top of the file that needs to be manually removed.)
commit 6a0fb2e1 (2023-01-10): Darwin: Add vendorID and productID properties to MTRDeviceAttestationDeviceInfo (#24335)

* Add Xcode file header templates

* Add vendorID and productID properties to MTRDeviceAttestationDeviceInfo

Fixes #23338
closed 2023-01-10 by andy31415