project-chip/connectedhomeip #23339: [Feature] Metering cluster

klem-everywhere, 2022-10-25:
### Feature description

```bash
Is power meter cluster in the scope of Matter?
I see that a lot of types are lacking from the page https://developers.home.google.com/matter/supported-devices
Is there a place where I could follow up plans, progress, RFCs?
```

### Platform

all

bzbarsky-apple, 2022-10-25:
https://github.com/CHIP-Specifications/connectedhomeip-spec/ is where spec development happens.

closed 2022-10-25 by bzbarsky-apple

lboue, 2022-10-25:
@bzbarsky-apple It seems that this is a private repo. We are getting a 404 not found error: 
https://github.com/CHIP-Specifications/connectedhomeip-spec/

clapre, 2022-10-26:
@lboue @klem-everywhere - The Matter Specification is available from a link here:
https://csa-iot.org/all-solutions/matter/
The development of future versions of the specification is being done within the _member-driven_ Connectivity Standards Alliance per their IPR policy, in order to ensure RANDZ licensing for the specification. Future versions of the specification will be released, in conjunction with the code base being developed (under Apache2) here.
We do have a group of members looking in to energy management over Matter, broadly speaking, including power metering for devices.

klem-everywhere, 2022-10-26:
Thanks @clapre.
Does anyone know a better place to keep track of different Matter implementations than https://developers.home.google.com/matter/supported-devices?

brainstorm, 2022-10-28:
> Thanks @clapre. Does anyone know a better place to keep track of different Matter implementations than https://developers.home.google.com/matter/supported-devices?

https://csa-iot.org/csa-iot_products/

lboue, 2022-10-31:
How can we get access to that development spec?

brainstorm, 2022-11-01:
> How can we get access to that development spec?

TL;DR, you must pay US$20000 yearly for the privilege, details here:

https://twitter.com/braincode/status/1585941743312117761

That's CSA's version of "open standard"... I guess that the play of words is: the standard is open, but (participation on building) the spec isn't.

lboue, 2022-11-01:
Can you confirm Adopter Members can access to CHIP-Specifications?

https://csa-iot.org/become-member/#Membership

> **Adopter Members ($7,000 USD/yr)** use existing, **approved specifications** to build products. Adopters can access completed, approved standards documents, certify products through Alliance certification programs, and use Alliance technology logos and trademarks for certified products.