project-chip/connectedhomeip #23342: [iOS] identify is always sent to bridge root endpoint, instead of bridged device endpoint

plan44, 2022-10-25:
### Reproduction steps

```bash
1. commission a bridge which contains one or multiple bridged devices (e.g. the `bridge-app` example)
2. during commissioning, Home.app UI walks though all bridged devices and offers a "identify" device button. 
3. pressing "identify" **always sends the Identify command (0 in cluster 3) to endpoint 0 (the root node) instead of to the bridged device** itself.

As the purpose of "identify" is to associate offered accessories (in the current implementation: anonymous because unnamed, see #22734) with actual hardware, this is clearly wrong. "identify" commands should be issued to the actual bridged device's endpoint so the corresponding hardware can make itself visible/audible and allow users to recognize them.

Note that sending "identify" to endpoint 0 is wrong even in the case of identifying the bridge itself, which is NOT represented by endpoint 0, but by a separate endpoint (1 in the `bridge-app` example).
```

### Platform

darwin

### Platform Version(s)

16.1 release

### Type

Manually tested with SDK

nivi-apple, 2022-10-27:
@plan44 Please file an issue via our feedback assistant and we should get this resolved for you.  Please find the instructions below. Please let me know if you need any help with filing one and I would be happy to help.

Thanks,
Nivi.

Feedback assistant ticket:

File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
Choose iOS/iPadOS to file an issue
Choose the area as Home App and fill in the details of the type of issue and the description
Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
iOS:
https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
tvOS:
https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.
Apple DeveloperApple Developer
[Bug Reporting - Apple Developer](https://developer.apple.com/bug-reporting/#feedback-assistant)
Now with Feedback Assistant available on iPhone, iPad, Mac, and the web, it’s easier to submit effective bug reports and request enhancements to APIs and tools. (71 kB)
https://developer.apple.com/bug-reporting/#feedback-assistant
developer.apple.comdeveloper.apple.com
[Profiles and Logs - Bug Reporting - Apple Developer](https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit)
Get details on providing logs, reproducible test cases, and other information that will help us investigate and diagnose your reported issues.
developer.apple.comdeveloper.apple.com
[Profiles and Logs - Bug Reporting - Apple Developer](https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network)
Get details on providing logs, reproducible test cases, and other information that will help us investigate and diagnose your reported issues.
developer.apple.comdeveloper.apple.com
[Profiles and Logs - Bug Reporting - Apple Developer](https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns)
Get details on providing logs, reproducible test cases, and other information that will help us investigate and diagnose your reported issues.

plan44, 2022-10-28:
Recent testing with iOS 16.2 beta shows that this issue is already fixed. So closing this issue.

@nivi-apple Thanks for the instructions for feedback assistant. I will post future non-SDK iOS issues there, and hope it will be less of a black hole experience when reporting matter/home related things than it used to be for other topics (I've been developing iOS apps since 2009...) 😉 😁

closed 2022-10-28 by plan44