project-chip/connectedhomeip #23353: [BUG] CHIPTool App doesn't work in iOS 16.1

zzz2016, 2022-10-26:
### Reproduction steps

```bash
1. Scan the QR code
2. Enter the Wifi info
3. Pair the device
```

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

26a054c31144478e3c27391c06b0dc3230bb0882

### Platform

darwin

### Platform Version(s)

16.1

### Anything else?

[image]

bzbarsky-apple, 2022-10-26:
@zzz2016 I believe you need to install https://developer.apple.com/services-account/download?path=/iOS/iOS_Logs/EnableBluetoothCentralMatterClientDeveloperMode.mobileconfig on iOS and turn on developer mode to use the test iOS CHIPTool.

closed 2022-10-26 by bzbarsky-apple

TheWalkingDead1024, 2022-10-27:
> @zzz2016 I believe you need to install https://developer.apple.com/services-account/download?path=/iOS/iOS_Logs/EnableBluetoothCentralMatterClientDeveloperMode.mobileconfig on iOS and turn on developer mode to use the test iOS CHIPTool.

@bzbarsky-apple  Many thanks for your replay. We also met this issue when we use matter framework in iOS SDK. Could you please tell us when it will be released in iOS.  It will be very helpful if you share us the milestone. Thanks again.

bzbarsky-apple, 2022-10-27:
@TheWalkingDead1024 If you are trying to use the system-provided Matter framework to commission devices, you should look into the MatterSupport framework.   See https://developer.apple.com/apple-home/matter/

TheWalkingDead1024, 2022-10-27:
> https://developer.apple.com/apple-home/matter/

@bzbarsky-apple Thanks again. Let's go back to the chip Matter framework(non system-provided). Would you please share us more information about when this issue will be resolved or, let me put it straight, when CHIPTool(chip framework) will work in iOS 16 without installing the profile and turn on developer mode.

huchuankan, 2024-03-14:
Matter Commission  demo video ： https://www.bilibili.com/video/BV1Nx421y7sa/