project-chip/connectedhomeip #23355: [Documentation] Commissioning failed when use PAI which pathlen is 0

KaMeHaMeHaaaa, 2022-10-26:
### Documentation issues

```bash
1. I generate A PAI with my own PAA, and command is:
 ./chip_cert gen-att-cert --type i --subject-cn "TP-Link Matter PAI" --subject-vid 1188 --subject-pid 0103 --valid-from "2021-06-28 14:23:43" --lifetime "4294967295" --error-type ext-basic-pathlen0 --ca-key ./paa_cert/paa_key.pem --ca-cert ./paa_cert/paa_cert.pem --out-key ./pai_cert/pai_key.pem --out ./pai_cert/pai_cert.pem

2. I generate A DAC with PAI in step 1, and command is:
./chip_cert gen-att-cert --type d --subject-cn "Matter DAC" --subject-vid 1188 --subject-pid 0103 --valid-from "2021-06-28 14:23:43" --lifetime "4294967295" --ca-key ./pai_cert/pai_key.pem --ca-cert ./pai_cert/pai_cert.pem --out-key ./dac_cert/dac_key.pem --out ./dac_cert/dac_cert.pem

3. Commissioning failed at AttenstationVerifaction, status code is 301

4. I genertae another PAI with only changing the pathlen from 0 to 1, then generate DAC with it, and commissioning is successful.
```

### Anything else?

[commissioning_log.txt](https://github.com/project-chip/connectedhomeip/files/9865542/commissioning_log.txt)
[DAC.txt](https://github.com/project-chip/connectedhomeip/files/9865547/DAC.txt)
[PAA.txt](https://github.com/project-chip/connectedhomeip/files/9865548/PAA.txt)
[PAI.txt](https://github.com/project-chip/connectedhomeip/files/9865549/PAI.txt)

KaMeHaMeHaaaa, 2022-10-26:
Hi, @tcarmelveilleux I uploaded the PAA PAI DAC I used, please have a look, thanks!

tcarmelveilleux, 2022-10-26:
You should not use `--error-type` at all. These are made to force non-standard behavior.

Also, what version of SDK is this code running against, both on commissioner chip-tool, and on device side?

tcarmelveilleux, 2022-10-26:
Can you please share the development CD you use?

emargolis, 2022-10-26:
The issue is unrelated to the pathlen constraint. Actually,  pathlen = 0 is the correct value for the PAI cert. 

The issue is validity time of the certificate: notBeforeDAC < notBeforePAI, which implies that PAI wasn't valid at the time of DAC issuance.

KaMeHaMeHaaaa, 2022-10-26:
Sorry, I forgot to mention that the PAA I used is generated by A chip-cert tool which built in an old SDK Version. (probably before TE9), and the PAI & DAC I used is generated by A chip-cert tool which built in SDK SVE2 Branch.  Use the old version chip-cert to generate PAI, it has a default pathlen value 1 (without add param --error-type),  and commissioning can complete successfully. But use the SVE2 version chip-cert to generate PAI, it has a default pathlen value 0(without add param --error-type), and commissioning failed with the old version chip-cert generated PAA.

KaMeHaMeHaaaa, 2022-10-26:
Here is the CD printed during commissioning
[1666746426.476222][92161:92167] CHIP:DMG: Attestation Elements (280 bytes) =
[1666746426.476278][92161:92167] CHIP:DMG: {
[1666746426.476324][92161:92167] CHIP:DMG:     Certification Declaration =
[1666746426.476369][92161:92167] CHIP:DMG:     {
[1666746426.478350][92161:92167] CHIP:DMG:         Format Version = 1
[1666746426.478431][92161:92167] CHIP:DMG:         Vendor Id = 4488
[1666746426.478486][92161:92167] CHIP:DMG:         Product Ids =
[1666746426.478534][92161:92167] CHIP:DMG:         {
[1666746426.478585][92161:92167] CHIP:DMG:             Product Id = 259
[1666746426.478636][92161:92167] CHIP:DMG:         }
[1666746426.478689][92161:92167] CHIP:DMG:         Device Type Id = 266
[1666746426.478747][92161:92167] CHIP:DMG:         Certificate Id (19) = ZIG20142ZB330003-24
[1666746426.478804][92161:92167] CHIP:DMG:         Security Level = 0
[1666746426.478856][92161:92167] CHIP:DMG:         Security Information = 0
[1666746426.478907][92161:92167] CHIP:DMG:         Version Number = 9876
[1666746426.478983][92161:92167] CHIP:DMG:         Certification Type = 1
[1666746426.479035][92161:92167] CHIP:DMG:     }
[1666746426.479090][92161:92167] CHIP:DMG:     Attestation Nonce (32) = 20C000C9B746AEF43F950065E208F4CBED734AB15EE407C4DEDF3C48D971BBC5
[1666746426.479139][92161:92167] CHIP:DMG:     Timestamp = 0
[1666746426.479185][92161:92167] CHIP:DMG: }

KaMeHaMeHaaaa, 2022-10-26:
I'm not sure whether  "the PAA generated by the old version chip-cert +  PAI & DAC generated by the SVE2 version chip-cert" will cause the failure of commissioning(without add param --error-type).

KaMeHaMeHaaaa, 2022-10-26:
> yes, I will try again , thanks

KaMeHaMeHaaaa, 2022-10-26:
> The issue is unrelated to the pathlen constraint. Actually, pathlen = 0 is the correct value for the PAI cert.
> 
> The issue is validity time of the certificate: notBeforeDAC < notBeforePAI, which implies that PAI wasn't valid at the time of DAC issuance.

Ok, I will try again, and thanks!

KaMeHaMeHaaaa, 2022-10-26:
I changed the validity time of DAC, make it bigger than PAI, and it works. Problem solved! Thanks for the help! @emargolis @tcarmelveilleux

closed 2022-10-26 by KaMeHaMeHaaaa