project-chip/connectedhomeip #23358: [Documentation] chip-tool - Subscribe interaction

sigidagi, 2022-10-26:
### Documentation issues

```bash
1. [chip-tool documentation on github](https://github.com/project-chip/connectedhomeip/tree/master/examples/chip-tool) needs some update regarding `temperaturemeasurement` cluster. from version 1.0 it has commands:

sigis:src$ chip-tool temperaturemeasurement
[1666782438.446900][4190997:4190997] CHIP:TOO: Missing command name
Usage:
  chip-tool temperaturemeasurement command_name [param1 param2 ...]

  +-------------------------------------------------------------------------------------+
  | Commands:                                                                           |
  +-------------------------------------------------------------------------------------+
  | * command-by-id                                                                     |
  | * read-by-id                                                                        |
  | * read                                                                              |
  | * write-by-id                                                                       |
  | * subscribe-by-id                                                                   |
  | * subscribe                                                                         |
  | * read-event-by-id                                                                  |
  | * subscribe-event-by-id                                                             |
  +-------------------------------------------------------------------------------------+

```

2. It needs clarification for `subscribe` command for `temperaturemeasurement` cluster, i.e `chip-tool temperaturemeasurement subscribe measured-value <min-interval> <max-interval> <destination-id> <endpoint-ids>` 

3. It needs clarification for `subscribe` command for `binding` cluster, i.e.  ` chip-tool binding subscribe binding <min-interval> <max-interval> <destination-id> <endpoint-ids>`

According Matter specification: "8.5 Subscribe Interaction" **subscriber** creates a subscription with a **publisher** If this interaction are between two nodes, it is not clear from (2) and (3) where these two nodes are provided as parameters.

My initial guess is that this relationship is provided in binding table, by executing:
```
chip-tool binding write binding '[{"fabricIndex": 1, "node": <id-of-subscriber>, "endpoint": 1, "cluster": 1026}]' <id-of-publisher> 1
```
Can you correct me?
```

ReneJosefsen, 2022-10-26:
Just to make sure I understand what you are trying to archive, are you trying to subscribe from chip-tool to measured-value on a device that has temperaturemeasurement implemented? In that case I would argue that you are not required to setup a binding, since chip-tool already knows the nodeID of the device?

Binding would be used for informing one device (which does not have any knowledge of an other devices on the network) ex. a switch where it can find another device, ex. a light.

One thing is is setup the binding, but also keep in mind that ACL also needs to be configured to allow for communication between the two devices. Again, in this case, if you want to setup the subscription from chip-tool, then it should already have an ACL entry to allow this.

sigidagi, 2022-10-26:
Hi @ReneJosefsen,

I have node that implements `temperaturemeasurement` server cluster and Linux node/application that is interested in displaying those measurements. Linux node should receive those temperature measurements at regular intervals, i.e. every 10 sec. It is not an option to use chip-tool to pull those measurements, i.e:
`chip-tool temperaturemeasurement read measured-value 52 1` 
then parse response and send those values to Linux matter application, which will display that value. 

I guess subscription should help for direct communication: node to node. In principal I can have more than one node with `temperaturemesurement` server cluster and more then one node for displaying those measurements. So I need example how to create relationship between concrete nodes (subscriber and publisher)

ACL is another issue.

sigidagi, 2022-10-26:
BTW more detailed question regarding subscription on (devzone)[https://devzone.nordicsemi.com/f/nordic-q-a/93276/thingy-53-matter---temperature-subscription]

ReneJosefsen, 2022-10-26:
The light-switch examples might be a good resource: https://github.com/project-chip/connectedhomeip/tree/master/examples/light-switch-app

sigidagi, 2022-10-27:
Thanks @ReneJosefsen,

I've implemented functionality of [light-switch](https://github.com/project-chip/connectedhomeip/tree/master/examples/light-switch-app) on my nrf52840dk development board and I have used `chip-tool` for binding, i.e in my case:
```
chip-tool binding write binding '[{"fabricIndex": 1, "node": 32, "endpoint": 1, "cluster": 6}]' 52 1
```
where `6` is OnOff cluster, `32` - linux matter app, `52` - nrf52840dk lighting-app with `temperaturemeasurement` and OnOff clusters. After binding and enabling ACL, I can press switch button - I get immediate response (On Off) on Linux app. So far so good.
But... 
OnOff cluster has `Commands` like *On*, *Off*, *Toggle*, and all logic is based on those commands in `BindingHandler` class
```
Controller::InvokeCommandRequest(....)
```
Temperaturemeasurement cluster do not provide any commands, so I can not use `Controller::InvokeCommandRequest`, so maybe such binding approach do not work. 

Problem description became too technical... :-) Can you provide link to forum, where I can ask specific technical questions regarding Matter?

ReneJosefsen, 2022-10-27:
InvokeCommandRequest is as you have noticed, used to invoke a command. There are similar commands to read/write or subscribe to attributes. Have a look at `Controller::SubscribeAttribute` and/or `Controller::ReadAttribute`.

I do not think there is a public "forum" for topics/discussions like this unfortunately.

sigidagi, 2022-11-09:
Hi @ReneJosefsen ,
bwt, I was able to build Linux Matter application and use `Controller::ReadAttribute` for fetching temperature measurements. Right now I have hardcoded  peer node ID (with server `temperaturemeasurement` cluster), but in principal it should be provided at runtime, let say by `chip-tool`. Does binding table will fit best for this purpose?
```
chip-tool binding write binding '[{"fabricIndex": 1, "node": 52, "endpoint": 1, "cluster": 1026}]' 32 1
```
where `52` is a peer node ID,  `32` Linux matter application

closed 2022-11-10 by sigidagi

ReneJosefsen, 2022-11-18:
Sorry for not getting back, it got lost I my mail box. But yes that looks about right. 👍🏻