project-chip/connectedhomeip #23360: [BUG] minimalMdns repeats announcements in violation of RFC6762

tcarmelveilleux, 2022-10-26:
### Reproduction steps

```bash
1.
2.
3.
...
```

### Bug prevalence

100%

### GitHub hash of the SDK that was being used

9c3fedab6c7fcd7a6856782ffea31ee301aec91a

### Platform

core

andy31415, 2022-10-31:
I am trying to reproduce this. Currently I am checking using minmdns examples (server and client) on linux since their behaviour should be the same as embedded and it is faster to debug. What I see is:

- On multicast queries only a single reply is sent. Everything else is throttled and not replied to even though the server considers it
- Unicast queries seem to be responded to many times: client attemts a query for every interface + ip address type combination, server seems to reply a LOT more. I believe a single broadcast is received many times: I changed the code to send a single broadcast and I received 8 replies on my network (I have multiple network cards and ipv4/ipv6 addresses on each).

andy31415, 2022-10-31:
I believe the issue we have is that the code attempts to listen on multiple interfaces, however a single listen should be sufficient: one for IPv4 and one for IPv6. Otherwise even though we attempt to bind UDP endpoints to specific interfaces, we will still receive all queries over every listen interface.

closed 2023-05-18 by stale[bot]