project-chip/connectedhomeip #23365: [BUG] Console not responding to input ESP32 example

ar0311, 2022-10-27:
### Reproduction steps

```bash
1. Build lighting-app ESP32 example
2. idf.py monitor
3. Device output displays fine but not responding to any input
...
```

### Bug prevalence

Everytime i build

### GitHub hash of the SDK that was being used

bd2c96a1944f3eb8e0e2497e9d2fa4747848d501

### Platform

esp32

ar0311, 2022-10-27:
Fixed, the shell enable option was not set in the default config.

Is this something that should be turned on by default?

bzbarsky-apple, 2022-10-31:
@dhrishi @shubhamdp

dhrishi, 2022-11-01:
@ar0311 I think it should be okay to enable the console by default. We will check and update

PR #23417 (2022-11-02): [ESP32] Enabled chip shell for lighting app
Problem

>  ENABLE_CHIP_SHELL config is disabled by default for lighting app
>  Fixes #23365

Testing
> Tested CLI for lighting app
closed 2022-11-02 by andy31415