project-chip/connectedhomeip #23368: [Feature] Does Apple Home APP support HVAC/thermostat module?

tobbe2007, 2022-10-27:
### Feature description

```bash
When I change the device type to HVAC(0x300) and thermostat(0x301)，the Home APP can commission with our device succ, but Home APP shows "This module isn't supported".

IOS version 16.1.
```

### Platform

all

### Anything else?

@bzbarsky-apple  Can you help me to check this feauture in Home APP?

bzbarsky-apple, 2022-10-27:
@nivi-apple

nivi-apple, 2022-10-27:
Please file a feedback assisstant ticket and we should be able to help. Details below. Please let me know if you need help with any of the steps below. Happy to help.
Thanks,
Nivi.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.
Apple DeveloperApple Developer
[Bug Reporting - Apple Developer](https://developer.apple.com/bug-reporting/#feedback-assistant)
Now with Feedback Assistant available on iPhone, iPad, Mac, and the web, it’s easier to submit effective bug reports and request enhancements to APIs and tools. (71 kB)
https://developer.apple.com/bug-reporting/#feedback-assistant

developer.apple.comdeveloper.apple.com
[Profiles and Logs - Bug Reporting - Apple Developer](https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit)
Get details on providing logs, reproducible test cases, and other information that will help us investigate and diagnose your reported issues.
developer.apple.comdeveloper.apple.com
[Profiles and Logs - Bug Reporting - Apple Developer](https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network)
Get details on providing logs, reproducible test cases, and other information that will help us investigate and diagnose your reported issues.
developer.apple.comdeveloper.apple.com
[Profiles and Logs - Bug Reporting - Apple Developer](https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns)
Get details on providing logs, reproducible test cases, and other information that will help us investigate and diagnose your reported issues.

bzbarsky-apple, 2022-10-31:
Closing, not an SDK issue.

closed 2022-10-31 by bzbarsky-apple