project-chip/connectedhomeip #23374: When NotifyUpdateApplied is received, evict other sessions to that node

bzbarsky-apple, 2022-10-28:
Receipt of NotifyUpdateApplied indicates that the other side has applied an OTA update, and presumably rebooted.

This seems like a signal that we should evict all other sessions (as in, except the one we get the command on) to the relevant node, since the chance that it preserved sessions across OTA is pretty slim.

@mrjerryjohns @jtung-apple thoughts?

mrjerryjohns, 2022-10-28:
Yes, that's a good thing to do. While the balanced LRU scheme will in most cases, prioritize reaping that old session, it's possible that it may also reap an older session to another node which in fact, might still be valid and active, causing possible delays interacting with that perfectly functioning node.

mrjerryjohns, 2022-10-28:
I'm torn about doing this in the OTA provider cluster, since it is a policy decision somewhat. Perhaps it should be an opt-out, through some kind of define?

bzbarsky-apple, 2022-10-28:
> While the balanced LRU scheme will in most cases, prioritize reaping that old session

Right, that's part of it, but the other part is that this lets us detect subscription drop faster in the OTA restart case...

> Perhaps it should be an opt-out, through some kind of define?

Yes, we can certainly do that, or add a way for the OTA delegate to opt out (a virtual method that it can override, etc).  Thoughts on which of those would be better?

mrjerryjohns, 2022-10-28:
I'd definitely prefer the delegate, makes this policy decision be a more flexible runtime choice.