project-chip/connectedhomeip #23381: [1.0] Why does not support multiple commands?

nykh2010, 2022-10-28:
### Reproduction steps

```bash
in spec chapter 10.6.9, the InvokeRequestMessage's InvokeRequests Element is an array type, but CommandSender not support multiple commands? what's the reason?
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

tag v1.0.0

### Platform

vscode

### Type

Spec Compliance Issue

bzbarsky-apple, 2022-10-31:
@nykh2010 There were open issues about how it should work, and limited time before 1.0 for answering those questions and implementation.

See also spec section "2.12.1. Invoke Multiple Paths":

> Support for an Invoke Interaction with multiple paths or wildcard paths is provisional.

vivien-apple, 2022-12-16:
Just to make it clear, the `chip-tool` syntax for such a command would be:
```
$ ./out/debug/standalone/chip-tool onoff toggle 0x12344321 1,2
```

It does not work as of today because the stack does not supports sending commands to multiple endpoints.
Once it works in on the stack side I could implement `chip-tool` supports for it.

yunhanw-google, 2024-10-07:
batch command has been supported since 1.3, close this one, thanks

closed 2024-10-07 by yunhanw-google