project-chip/connectedhomeip #23384: Figure out whether we are mis-using instancetype in the Darwin APIs

bzbarsky-apple, 2022-10-28:
See https://github.com/project-chip/connectedhomeip/pull/23364#discussion_r1007588070 -- what does `instancetype` promise, exactly?  When is it appropriate to use it?

@jtung-apple @ksperling-apple

bzbarsky-apple, 2022-10-28:
Also see https://github.com/project-chip/connectedhomeip/pull/23364#discussion_r1007588185

jtung-apple, 2023-02-16:
Quoting @ksperling-apple - first comment:
> Not changed in this PR, but it looks to me like the return type should be `MTRDevice *` instead of `instancetype`? If you were to call this method on a sub-class of `MTRDevice`, would you be guaranteed to get an instance of that sub-class?

A subclass of `MTRDevice` would return an object of the subclass in "reasonable code", as in, using the class names explicitly.

If you play with a `Class` variable in a way that limits the compiler's ability to infer the actual class, then you may get an `id` which is assignable to anything (the runtime type will still be correct, if you ask the object for its class). But doing this is not good practice in ordinary code.

bzbarsky-apple, 2023-02-16:
I think the issue is this: for a _factory method_ as here, what happens?  It can do one of two things.  The impl could `[[MTRDevice alloc] init]` as I think we do, or it could `[[self.class alloc] init]`, right?  And it seems like we should return `instancetype` in the latter case and `MTRDevice *` in the former?

jtung-apple, 2023-02-16:
Okay you're right about this. Given the implementation of the factory method asks `MTRDeviceController` for a `MTRDevice` object, the return type really can't be anything but `MTRDevice *`. And subclassing MTRDevice really doesn't make sense because of that.

jtung-apple, 2023-02-16:
We should change the factory method to return `MTRDevice *`.

closed 2023-02-17 by bzbarsky-apple