project-chip/connectedhomeip #23394: [PSoC6]How do I remove commissoning data?

mmt-10, 2022-10-31:
### Reproduction steps

```bash
for PSoC6
$ cd ~/connectedhomeip
$ source scripts/activate.sh
$ scripts/examples/gn_psoc6_example.sh
$ scripts/build/build_examples.py --no-log-timestamps --target 'infineon-psoc6-lock' build
$ python3 out/infineon-psoc6-lock/chip-psoc6-lock-example.flash.py
$ cu -s 115200 -l /dev/ttyACM0

chip-tool
$ ./chip-tool pairing ble-wifi 12344321 $SSID $PASS 20202021 3840 --ble-adapter 1
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

commit 7f2b0a5920d1a7acc202c2932ac4c626c54cd31e (HEAD -> v1.0-branch, origin/v1.0-branch)

### Platform

other

### Anything else?

I would like to remove commissioning data from PSoC6.
because I would like to update WiFi AP (SSID & PASSWORD).

I rewrite firmware by below command, however, WiFi AP data was not cleared.

> $ python3 out/infineon-psoc6-lock/chip-psoc6-lock-example.flash.py

And, I retry commissioning and it failed because BLE on CY8CKIT-062S2-43012 stopped.
How do I retry commissioning?

### Environment
hardware: [CY8CKIT-062S2-43012](https://www.cypress.com/CY8CKIT-062S2-43012)
example: lock-app
[https://github.com/project-chip/connectedhomeip/tree/v1.0-branch/examples/lock-app/infineon/psoc6](https://github.com/project-chip/connectedhomeip/tree/v1.0-branch/examples/lock-app/infineon/psoc6)

bzbarsky-apple, 2022-10-31:
@praveenCY Could you take a look please?

praveenCY, 2022-10-31:
@mmt-10 @bzbarsky-apple : Please press USER_BTN2 on the CY8CKIT-062S2-43012 to factory reset (This will clear all the commissioning data from PSoC6 board)

mmt-10, 2022-10-31:
@praveenCY 
Thank you for your reply.
I could clear all commissioning data from PSoC6.

I tried only long press before.
I understand USER_BTN2 needs 2 actions which click (short press) and long press over 5sec.

closed 2022-10-31 by mmt-10