project-chip/connectedhomeip #23397: [BUG] [iOS]Is matter device must be paired from HomeKit？

JourneyYoung, 2022-10-31:
### Reproduction steps

```bash
1.pair device from qrcode 
2.BLE:Error writing Characteristics in Chip service on the device: [The specified UUID is not allowed for this operation.]
3.then failed
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

master

### Platform

darwin

### Platform Version(s)

16.1

### Anything else?

Can we find matter device bluetooth service after we certified by the CSA？ or we must contect device from homekit after iOS 16.1 ?

bzbarsky-apple, 2022-11-30:
@JourneyYoung after 16.1 you must use the MatterSupport framework to pair devices.

bzbarsky-apple, 2022-11-30:
@JourneyYoung See https://developer.apple.com/documentation/mattersupport

closed 2022-11-30 by bzbarsky-apple

geoffroyletourneur, 2022-12-14:
No. Non-developer systems will not be able to write that UUID. The profile is there for testing purposes only; production systems should be using the MatterSupport framework to commission devices over BLE.

bzbarsky-apple, 2022-12-17:
> Do we need to have a specific app certificate or profile to enable DNS-SD on matter tcp/udp port on iOS 16.1/16.1.1?

I don't believe so...

>  MatterSwiftPoc[6406:1510181] MTRDeviceAttestationDelegateBridge::OnDeviceAttestationFailed completed with result: 601

That's `kCertificationDeclarationNoCertificateFound`.  What does the certification declaration certificate on the device look like?  What key is it signed with?

uascent, 2022-12-22:
> https://developer.apple.com/news/?id=0oi77447

 I had used the Matter Support to Add the Matter Devices, when pairing the device, everything is okay, and the device be commissinged successfully , but finally, there is an dialog which HMErrorCode: 2, What's that mean Error Code: 2

bzbarsky-apple, 2022-12-22:
@uascent that looks like `HMErrorCodeNotFound`.  What that _means_ in this case (i.e. what was not found), though...

uascent, 2022-12-23:
> @uascent that looks like `HMErrorCodeNotFound`. What that _means_ in this case (i.e. what was not found), though...

Don't know what happen, log didn't show more tips, so could you provide the sample code about how to use the matter support framework commission successfully;  what tips should I know? thanks; 

[image]

Code:

[image]

bzbarsky-apple, 2023-01-03:
@uascent You should probably file a Feedback Assistant ticket for that:

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.
Apple DeveloperApple Developer
[Bug Reporting - Apple Developer](https://developer.apple.com/bug-reporting/#feedback-assistant)
Now with Feedback Assistant available on iPhone, iPad, Mac, and the web, it’s easier to submit effective bug reports and request enhancements to APIs and tools. (71 kB)
https://developer.apple.com/bug-reporting/#feedback-assistant

huchuankan, 2024-03-14:
NO need

huchuankan, 2024-03-14:
Matter Commission  demo video ： https://www.bilibili.com/video/BV1Nx421y7sa/