project-chip/connectedhomeip #23400: SDK feature implementation needed for V1.1

Survensa, 2022-10-31:
### Feature description

```bash
The following features have to be implemented for V1.1

1. Group Key Management Cluster: CacheandSync Feature
2. Power Source Cluster: Wired, Rechargeable, and Replaceable
3. Time Format Localization Cluster: CalendarFormat
4. Level Control: Frequency
5. Switch: Momentary Switch release, Long Press, multi-press
6. Pressure Measurement:  Extended (Optional Feature)
7. Door Lock: Finger Credentials and Face Credentials
```

### Platform

all

bzbarsky-apple, 2022-10-31:
This probably needs one issue per item.  For example, item 7 is covered by https://github.com/project-chip/connectedhomeip/issues/21656

Survensa, 2022-11-01:
@bzbarsky-apple Will do the same and close this