project-chip/connectedhomeip #23412: [BUG] ninja -C out/host check is blocked

juanariaspriceline, 2022-10-31:
### Reproduction steps

```bash
When I ran this commmand I got:

ninja: Entering directory `out/host'
[0/5] ACTION //src/platform/tests:TestDnssd_run(//build/toolchain/host:mac_arm64_gcc)
```
```

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

eeb4cdcd962f64a9ffab8c16d537031add717661

### Platform

darwin

juanariaspriceline, 2022-10-31:
MacOS Ventura 13.0

bzbarsky-apple, 2022-11-01:
I have encountered this if I have an iPhone plugged into the laptop via USB: we advertise on that USB interface and then expect to be able to see that advertisement, but that part fails.

If that's not your situation, what happens if you kill the build and run the `TestDnssd` binary (which lives somewhere under `out`) by hand?  Can you debug why the test is not completing?

woody-apple, 2023-06-23:
Closing, as there has been no activity in a while.

closed 2023-06-23 by woody-apple