project-chip/connectedhomeip #23420: [1.0] ninja: error: '../../../../../../../../../../../../../Volumes/Data', needed by 'obj/src/app/libCHIPDataModel.AttributeAccessInterface.cpp.o', missing and no known rule to make it

aelposhi, 2022-11-01:
### Reproduction steps

```bash
+ mkdir -p /Users/aelposhi/Library/Developer/Xcode/DerivedData/Darwin-fiarwprtkzhzakagiaewjawagrju/Build/Intermediates.noindex/Matter.build/Debug-maccatalyst/Matter.build
+ export
+ defines=()
+ declare -a defines
+ read -r -a defines
+ declare target_defines=
+ for define in '"${defines[@]}"'
+ case "$define" in
+ continue
+ for define in '"${defines[@]}"'
+ case "$define" in
+ target_defines+=',"CHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=<lib/address_resolve/AddressResolve_DefaultImpl.h>"'
+ for define in '"${defines[@]}"'
+ case "$define" in
+ target_defines+=',"DEBUG=1"'
+ for define in '"${defines[@]}"'
+ case "$define" in
+ target_defines+=',"CHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=<lib/address_resolve/AddressResolve_DefaultImpl.h>"'
+ for define in '"${defines[@]}"'
+ case "$define" in
+ target_defines+=',"MTR_NEWLY_AVAILABLE="'
+ for define in '"${defines[@]}"'
+ case "$define" in
+ target_defines+=',"MTR_NEWLY_DEPRECATED(message)="'
+ target_defines='["CHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=<lib/address_resolve/AddressResolve_DefaultImpl.h>","DEBUG=1","CHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=<lib/address_resolve/AddressResolve_DefaultImpl.h>","MTR_NEWLY_AVAILABLE=","MTR_NEWLY_DEPRECATED(message)="]'
+ declare target_arch=
+ declare target_cpu=
+ declare target_cflags=
+ read -r -a archs
+ for arch in '"${archs[@]}"'
+ '[' -z ']' ']'
+ '[' -n '' ']'
+ target_cflags+='"-arch","x86_64"'
+ [[ NO == YES ]]
+ args=('default_configs_cosmetic=[]' 'chip_crypto="boringssl"' 'chip_build_tools=false' 'chip_build_tests=false' 'chip_enable_wifi=false' 'chip_log_message_max_size=4096' 'chip_disable_platform_kvs=true' "target_cpu=\"$target_cpu\"" "target_defines=$target_defines" "target_cflags=[$target_cflags]" "mac_target_arch=\"$target_arch\"" "mac_deployment_target=\"$LLVM_TARGET_TRIPLE_OS_VERSION$LLVM_TARGET_TRIPLE_SUFFIX\"")
+ declare -a args
+ [[ Debug == Debug ]]
+ args+=('is_debug=true')
+ [[ macOS != macOS ]]
+ [[ macOS == macOS ]]
+ args+=('target_os="mac"')
+ cd '/Volumes/Data Disk/Matter/ConnectedHomeIP/connectedhomeip'
++ find_in_ancestors chip_xcode_build_connector_env.sh
+ ENV=
+ [[ -n '' ]]
+ scripts/checkout_submodules.py --shallow --platform darwin
Checking out: nlassert, nlfaultinjection, nlio, nlunit-test, mbedtls, qrcode, pigweed, openthread, nanopb, zap, third_party/pybind11/repo, third_party/jsoncpp/repo, editline, third_party/boringssl/repo/src, open-iot-sdk-storage
+ [[ -z '' ]]
+ env -i PW_ENVSETUP_NO_BANNER=1 PW_ENVSETUP_QUIET=1 bash -c '. scripts/activate.sh'
20221101 11:08:12 WRN Current uname (22.1.0 Darwin Kernel Version 22.1.0: Sun Oct  9 20:15:09 PDT 2022; root:xnu-8792.41.9~2/RELEASE_ARM64_T6000 arm64) does not match Bootstrap uname (22.1.0 Darwin Kernel Version 22.1.0: Sun Oct  9 20:15:09 PDT 2022; root:xnu-8792.41.9~2/RELEASE_ARM64_T6000 arm64), you may need to rerun bootstrap on this system
+ set +ex
20221101 11:08:12 WRN Current uname (22.1.0 Darwin Kernel Version 22.1.0: Sun Oct  9 20:15:09 PDT 2022; root:xnu-8792.41.9~2/RELEASE_ARM64_T6000 arm64) does not match Bootstrap uname (22.1.0 Darwin Kernel Version 22.1.0: Sun Oct  9 20:15:09 PDT 2022; root:xnu-8792.41.9~2/RELEASE_ARM64_T6000 arm64), you may need to rerun bootstrap on this system
+ cd /Users/aelposhi/Library/Developer/Xcode/DerivedData/Darwin-fiarwprtkzhzakagiaewjawagrju/Build/Intermediates.noindex/Matter.build/Debug-maccatalyst/Matter.build
+ gn '--root=/Volumes/Data Disk/Matter/ConnectedHomeIP/connectedhomeip' gen --check out '--args=default_configs_cosmetic=[] chip_crypto="boringssl" chip_build_tools=false chip_build_tests=false chip_enable_wifi=false chip_log_message_max_size=4096 chip_disable_platform_kvs=true target_cpu="" target_defines=["CHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=<lib/address_resolve/AddressResolve_DefaultImpl.h>","DEBUG=1","CHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=<lib/address_resolve/AddressResolve_DefaultImpl.h>","MTR_NEWLY_AVAILABLE=","MTR_NEWLY_DEPRECATED(message)="] target_cflags=["-arch","x86_64"] mac_target_arch="" mac_deployment_target="ios13.4-macabi" is_debug=true target_os="mac"'
Done. Made 4767 targets from 221 files in 346ms
+ exec ninja -v -C out
ninja: Entering directory `out'
ninja: error: '../../../../../../../../../../../../../Volumes/Data', needed by 'obj/src/app/libCHIPDataModel.AttributeAccessInterface.cpp.o', missing and no known rule to make it
```

### Bug prevalence

Every time

### GitHub hash of the SDK that was being used

Drawin

### Platform

darwin

### Platform Version(s)

1.0

### Type

Platform Issue

### Anything else?

Hello,
While compiling the IOS mobile app in MACOS M1 I got the above error.
Can anyone advise, please?

bzbarsky-apple, 2022-11-01:
Compare:

```
 gn '--root=/Volumes/Data Disk/Matter/ConnectedHomeIP/connectedhomeip'
```

and

```
ninja: error: '../../../../../../../../../../../../../Volumes/Data'
```

Something is not handling the path with a space in it correctly...  @mspang @andy31415 

In the meantime, I expect putting `connectedhomeip` somewhere that does not have spaces in the path will work.

andy31415, 2022-11-01:
I think there is a good chance chip scripts in general to not handle spaces in names well. Maintaining such a support is unlikely to gain high priority. I suggest using some symlinks/names without spaces.

aelposhi, 2022-11-01:
> I think there is a good chance chip scripts in general to not handle spaces in names well. Maintaining such a support is unlikely to gain high priority. I suggest using some symlinks/names without spaces.

I respect your comment regarding the priority, but I still can't compile the APP.

mspang, 2022-11-01:
Our scripts should actually be OK, it's GN that's buggy. The wrong escaping mode is used in two places.

This patch to GN seems to be sufficient

```
diff --git a/src/gn/ninja_action_target_writer.cc b/src/gn/ninja_action_target_writer.cc
index 60d4204d..01bfb2cd 100644
--- a/src/gn/ninja_action_target_writer.cc
+++ b/src/gn/ninja_action_target_writer.cc
@@ -141,10 +141,16 @@ std::string NinjaActionTargetWriter::WriteRuleDefinition() {
     out_ << std::endl;
   }
 
+  // The command line requires shell escaping to properly handle filenames
+  // with spaces.
+  PathOutput command_output(path_output_.current_dir(),
+                            settings_->build_settings()->root_path_utf8(),
+                            ESCAPE_NINJA_COMMAND);
+
   out_ << "  command = ";
-  path_output_.WriteFile(out_, settings_->build_settings()->python_path());
+  command_output.WriteFile(out_, settings_->build_settings()->python_path());
   out_ << " ";
-  path_output_.WriteFile(out_, target_->action_values().script());
+  command_output.WriteFile(out_, target_->action_values().script());
   for (const auto& arg : args.list()) {
     out_ << " ";
     SubstitutionWriter::WriteWithNinjaVariables(arg, args_escape_options, out_);
diff --git a/src/gn/path_output.cc b/src/gn/path_output.cc
index d2886d7e..86c92e67 100644
--- a/src/gn/path_output.cc
+++ b/src/gn/path_output.cc
@@ -145,7 +145,7 @@ void PathOutput::WriteSourceRelativeString(std::ostream& out,
   } else {
     // Ninja (and none) escaping can avoid the intermediate string and
     // reprocessing of the inverse_current_dir_.
-    out << inverse_current_dir_;
+    EscapeStringToStream(out, inverse_current_dir_, options_);
     EscapeStringToStream(out, str, options_);
   }
 }
```

aelposhi, 2022-11-01:
> Our scripts should actually be OK, it's GN that's buggy. The wrong escaping mode is used in two places.
>
> This patch to GN seems to be sufficient
>
> ```
> diff --git [...]

Can you advise how to patch it, please?

mspang, 2022-11-02:
> Can you advise how to patch it, please?

You'd have to build from source, described at https://gn.googlesource.com/gn/#getting-a-binary

At least until such fix is landed & a new release cut.

aelposhi, 2022-11-06:
> You'd have to build from source, described at https://gn.googlesource.com/gn/#getting-a-binary
>
> At least until such fix is landed & a new release cut.

I'm compiling the APP using Xcode, as my plan is to compile the IOS app. So can you please advise how I can bypass the GN bug with Xcode?