project-chip/connectedhomeip #23435: [BUG] MacOS Ventura 13.0, CHIP-Tool cannot establish a BLE connection to pair devices into a thread network

benjaminnestler, 2022-11-02:
### Reproduction steps

```bash
After update of MacOS to Version V13.0 (Ventura) the chip-tool doesn't work as expected.

I tried to pair a device into a existing thread-network via ble.

The error is 

[1667391311686] [3120:29505] CHIP: [BLE] BLE:Error writing Characteristics in Chip service on the device: [The specified UUID is not allowed for this operation.]
[1667391311686] [3120:29505] CHIP: [IN] Failed to establish BLE connection: ../../../../../opt/nordic/ncs/v2.1.0/modules/lib/matter/examples/chip-tool/third_party/connectedhomeip/src/platform/Darwin/BleConnectionDelegateImpl.mm:260: Ble Error 0x00000407: GATT write characteristic operation failed
```

The pairing procedure works with MacOS V12.x

The whole log of chip-tool:

```
sudo ./chip-tool pairing ble-thread 0x100 hex:0e080000000000010000000300000f35060004001fffe0020811111111222222220708fdcb9feb8e974a17051000112233445566778899aabbccddeeff030e4f70656e54687265616444656d6f010212340410445f2b5ca6f2a93a55ce570a70efeecb0c0402a0f7f8 20202021 3840
Password:
[1667391310246] [3120:29499] CHIP: [DL] KVS will be written to: file:///tmp/chip_kvs
[1667391310290] [3120:29499] CHIP: [DL] _Init
[1667391310290] [3120:29499] CHIP: [IN] UDP::Init bind&listen port=0
[1667391310290] [3120:29499] CHIP: [IN] UDP::Init bound to port=53245
[1667391310290] [3120:29499] CHIP: [IN] UDP::Init bind&listen port=0
[1667391310290] [3120:29499] CHIP: [IN] UDP::Init bound to port=49626
[1667391310290] [3120:29499] CHIP: [IN] BLEBase::Init - setting/overriding transport
[1667391310290] [3120:29499] CHIP: [IN] TransportMgr initialized
[1667391310290] [3120:29499] CHIP: [FP] Initializing FabricTable from persistent storage
[1667391310290] [3120:29499] CHIP: [TS] Last Known Good Time: [unknown]
[1667391310290] [3120:29499] CHIP: [TS] Setting Last Known Good Time to firmware build time 2022-11-02T11:21:51
[1667391310292] [3120:29499] CHIP: [ZCL] Using ZAP configuration...
[1667391310292] [3120:29499] CHIP: [CTL] System State Initialized...
[1667391310292] [3120:29499] CHIP: [CTL] Stopping commissioning discovery over DNS-SD
[1667391310292] [3120:29499] CHIP: [CTL] Setting attestation nonce to random value
[1667391310292] [3120:29499] CHIP: [CTL] Setting CSR nonce to random value
[1667391310293] [3120:29499] CHIP: [IN] UDP::Init bind&listen port=5550
[1667391310293] [3120:29499] CHIP: [IN] UDP::Init bound to port=5550
[1667391310293] [3120:29499] CHIP: [IN] UDP::Init bind&listen port=5550
[1667391310293] [3120:29499] CHIP: [IN] UDP::Init bound to port=5550
[1667391310293] [3120:29499] CHIP: [IN] TransportMgr initialized
[1667391310293] [3120:29505] CHIP: [DL] _OnPlatformEvent
[1667391310293] [3120:29505] CHIP: [CTL] Stopping commissioning discovery over DNS-SD
[1667391310293] [3120:29505] CHIP: [CTL] Setting attestation nonce to random value
[1667391310293] [3120:29505] CHIP: [CTL] Setting CSR nonce to random value
[1667391310293] [3120:29505] CHIP: [CTL] Couldn't get ExampleOpCredsCAKey from storage: ../../../../../opt/nordic/ncs/v2.1.0/modules/lib/matter/examples/chip-tool/config/PersistentStorage.cpp:85: CHIP Error 0x000000A0: Value not found in the persisted storage
[1667391310294] [3120:29505] CHIP: [CTL] Couldn't get ExampleOpCredsICAKey from storage: ../../../../../opt/nordic/ncs/v2.1.0/modules/lib/matter/examples/chip-tool/config/PersistentStorage.cpp:85: CHIP Error 0x000000A0: Value not found in the persisted storage
[1667391310295] [3120:29505] CHIP: [CTL] Generating RCAC
[1667391310296] [3120:29505] CHIP: [CTL] Generating ICAC
[1667391310296] [3120:29505] CHIP: [CTL] Generating NOC
[1667391310297] [3120:29505] CHIP: [FP] Validating NOC chain
[1667391310298] [3120:29505] CHIP: [FP] NOC chain validation successful
[1667391310298] [3120:29505] CHIP: [FP] Added new fabric at index: 0x1
[1667391310298] [3120:29505] CHIP: [FP] Assigned compressed fabric ID: 0x8FD395E8960753E0, node ID: 0x000000000001B669
[1667391310298] [3120:29505] CHIP: [TS] Last Known Good Time: 2022-11-02T11:21:51
[1667391310298] [3120:29505] CHIP: [TS] New proposed Last Known Good Time: 2021-01-01T00:00:00
[1667391310298] [3120:29505] CHIP: [TS] Retaining current Last Known Good Time
[1667391310298] [3120:29505] CHIP: [FP] Metadata for Fabric 0x1 persisted to storage.
[1667391310299] [3120:29505] CHIP: [TS] Committing Last Known Good Time to storage: 2022-11-02T11:21:51
[1667391310300] [3120:29505] CHIP: [CTL] Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000
[1667391310300] [3120:29505] CHIP: [IN] UDP::Init bind&listen port=5550
[1667391310300] [3120:29505] CHIP: [IN] UDP::Init bound to port=5550
[1667391310300] [3120:29505] CHIP: [IN] UDP::Init bind&listen port=5550
[1667391310300] [3120:29505] CHIP: [IN] UDP::Init bound to port=5550
[1667391310300] [3120:29505] CHIP: [IN] TransportMgr initialized
[1667391310306] [3120:29505] CHIP: [BLE] NewConnection
[1667391310306] [3120:29505] CHIP: [BLE] CancelConnection
[1667391310307] [3120:29505] CHIP: [IN] SecureSession[0x600000394000]: Allocated Type:1 LSID:29034
[1667391310307] [3120:29505] CHIP: [SC] Assigned local session key ID 29034
[1667391310307] [3120:29505] CHIP: [SC] Including MRP parameters in PBKDF param request
[... 38 lines cut ...]
[1667391340321] [3120:29499] CHIP: [DMG] All ReadHandler-s are clean, clear GlobalDirtySet
[1667391340321] [3120:29499] CHIP: [BLE] CancelConnection
[1667391340323] [3120:29499] CHIP: [DL] Inet Layer shutdown
[1667391340323] [3120:29499] CHIP: [DL] BLE shutdown
[1667391340323] [3120:29499] CHIP: [DL] System Layer shutdown
[1667391340323] [3120:29499] CHIP: [TOO] Run command failure: ../../../../../opt/nordic/ncs/v2.1.0/modules/lib/matter/examples/chip-tool/third_party/connectedhomeip/src/protocols/secure_channel/PASESession.cpp:245: CHIP Error 0x00000032: Timeout
bennynestler@MBPvonBenjamin(x86_64) ~/workspaces/matter/chip-tool-110222 % 
```
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

561d23d

### Platform

nrf

### Platform Version(s)

2.1.0

bzbarsky-apple, 2022-11-04:
The profile mentioned in those issues is iOS-only.  MacOS needs a different profile, which may not exist yet.  I recall @cjandhyala running into this too...  @woody-apple

benjaminnestler, 2022-11-07:
Okay ... are you sure @bzbarsky-apple 

[image]

bzbarsky-apple, 2022-11-08:
> Okay ... are you sure @bzbarsky-apple

No, I am not (not least because the profile might have been updated since the last time I looked at it).  Try it and see whether it helps @benjaminnestler ?

benjaminnestler, 2022-11-08:
I've tried it and no it doesn't work. CHIP tool is not able to pair devices over BLE.

bzbarsky-apple, 2022-11-08:
@woody-apple

geoffroyletourneur, 2022-11-10:
@benjaminnestler  @bzbarsky-apple Did you see this trace ? I added some logs. 

I have the feeling BLE Core on mac OSX Ventura is bugged... 
We cannot called didWriteValueForCharacteristic on the CBService: 0x600001718a00, isPrimary = YES, UUID = FFF6

didWriteValueForCharacteristic: characteristic service:<CBService: 0x600001718a00, isPrimary = YES, UUID = FFF6>
didWriteValueForCharacteristic: characteristic descriptors:(null)
didWriteValueForCharacteristic: characteristic properties:8
didWriteValueForCharacteristic: characteristic isNotifying:0
didWriteValueForCharacteristic: characteristic isBroadcasted:0

benjaminnestler, 2022-11-14:
@geoffroyletourneur 

The error message of CHIP-Tool is

```
[1668410783126] [58984:407905] CHIP: [BLE] BLE:Error writing Characteristics in Chip service on the device: [The specified UUID is not allowed for this operation.]
[1668410783126] [58984:407905] CHIP: [IN] Failed to establish BLE connection: ../../../../../opt/nordic/ncs/v2.1.0/modules/lib/matter/examples/chip-tool/third_party/connectedhomeip/src/platform/Darwin/BleConnectionDelegateImpl.mm:260: Ble Error 0x00000407: GATT write characteristic operation failed
```

**I've updated my MacOS to V13.0.1 --> error still occurs**

I will test the CHIP-Tool from master-branch ...

bzbarsky-apple, 2022-11-18:
There is a devloper profile you have to install on new-enough iOS to allow this write characteristic.

There is no such profile available for MacOS Ventura yet, but once there is I will update this issue with a link to it.

Note that next week (Nov 21-25) is a holiday week at Apple, so if the profile does not appear on Nov 18 I would not expect it until Nov 28 at the earliest.

benjaminnestler, 2022-11-21:
Okay thank you @bzbarsky-apple

benjaminnestler, 2022-12-05:
Hi @bzbarsky-apple,
thanks for your response. The profile works with MacOS 13.1 beta 4.

Is the Bluetooth-Profile integrated in the next MacOS release-candidate?

bzbarsky-apple, 2022-12-08:
> Is the Bluetooth-Profile integrated in the next MacOS release-candidate?

No.  Non-developer systems will not be able to write that UUID.  The profile is there for testing purposes only; production systems should be using the MatterSupport framework to commission devices over BLE.

apro-michael, 2023-02-14:
After impot the mobileconfig file, After upgrade my MacOS(M1) from 13.0 to 13.2, it works. The error "BLE:Error writing Characteristics in Chip service on the device: [The specified UUID is not allowed for this operation.]" was gone.

woody-apple, 2023-06-23:
To learn more about how to test and develop on Apple devices, please refer to the documentation here: https://developer.apple.com/documentation/mattersupport/adding-matter-support-to-your-ecosystem

Developer mode is outlined there which can be used for testing.

woody-apple, 2023-06-23:
MatterSupport is also documented here: https://developer.apple.com/apple-home/matter/

closed 2023-06-23 by woody-apple

huchuankan, 2024-03-14:
do not need

Matter Commission  demo video ： https://www.bilibili.com/video/BV1Nx421y7sa/