project-chip/connectedhomeip #23489: [BUG] [PSoC6] [lock-app] LED is not reflect in chip-tool command

mmt-10, 2022-11-04:
### Reproduction steps

```bash
1. build and flash lock-app example for PSoC6 by following URL.
https://github.com/project-chip/connectedhomeip/tree/v1.0-branch/examples/lock-app/infineon/psoc6

2. commissioning from chip-tool on linux.
./chip-tool pairing ble-wifi 12344321 $SSID $PASS 20202021 3840 --ble-adapter 1

3. do on/off command
./chip-tool onoff on 12344321 1
./chip-tool onoff off 12344321 1

LED8 and LED9 were not working.

4. read on/off status
onoff status is followed above onoff command result.

./chip-tool onoff read on-off 12344321 1

5. LED8 and LED9 were working by pressing USER_BTN1
However,the result is now reflected in read onoff command.
for example,
 LED9 is emitting(status = lock) by USER_BTN1, however, the result of `chip-tool onoff read on-off` is `false`.
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

commit 7f2b0a5920d1a7acc202c2932ac4c626c54cd31e (HEAD -> v1.0-branch, origin/v1.0-branch)

### Platform

other

### Anything else?

the chip-tool command does not reflect to LED8/9.
set `onoff` to `on` by chip-tool on linux, but LED8/9 does not work and not emit.

when USER_BTN1 is pressed, LED8 and LED9 are working.
however, LED8/9 status is not reflect result of `chip-tool onoff read on-off`.
LED9 status is LOCK ( LED9 is emitting), but, `chip-tool onoff read on-off` is `false`.

USER_BTN1 and LED8/9 are chip-tool command independent.

How do I control LED8/9 from chip-tool command?

mmt-10, 2022-11-08:
@bzbarsky-apple 
I could control LED, but I don't know this solution is correct or not.

1. `USER_BTN1` -> `ZCL_ON_OFF_ATRRIBUTE`
[https://github.com/project-chip/connectedhomeip/blob/0b17bce8005fb155e78e243f31111cdaf2c92797/examples/lock-app/infineon/psoc6/src/AppTask.cpp#L651](https://github.com/project-chip/connectedhomeip/blob/0b17bce8005fb155e78e243f31111cdaf2c92797/examples/lock-app/infineon/psoc6/src/AppTask.cpp#L651)

```c
void AppTask::UpdateCluster(intptr_t context)
{
    uint8_t newOnOffState = !LockMgr().NextState();
    EmberAfStatus OnOffstatus =
        emberAfWriteAttribute(1, ZCL_ON_OFF_CLUSTER_ID, ZCL_ON_OFF_ATTRIBUTE_ID, (uint8_t *) &newOnOffState, ZCL_BOOLEAN_ATTRIBUTE_TYPE);

    if (OnOffstatus != EMBER_ZCL_STATUS_SUCCESS)
    {
        P6_LOG("ERR: updating lock state %x", OnOffstatus);
    }
}
```
2. `ZCL_ON_OFF_ATRRIBUTE` -> LED
[https://github.com/project-chip/connectedhomeip/blob/0b17bce8005fb155e78e243f31111cdaf2c92797/examples/lock-app/infineon/psoc6/src/ZclCallbacks.cpp#L37](https://github.com/project-chip/connectedhomeip/blob/0b17bce8005fb155e78e243f31111cdaf2c92797/examples/lock-app/infineon/psoc6/src/ZclCallbacks.cpp#L37)

```c
void MatterPostAttributeChangeCallback(const chip::app::ConcreteAttributePath & attributePath, uint8_t type, uint16_t size,
                                       uint8_t * value)
{
    EndpointId endpointId   = attributePath.mEndpointId;
    ClusterId clusterId     = attributePath.mClusterId;
    AttributeId attributeId = attributePath.mAttributeId;
    ChipLogProgress(Zcl, "Cluster callback: " ChipLogFormatMEI, ChipLogValueMEI(clusterId));

    if (clusterId == DoorLock::Id && attributeId == DoorLock::Attributes::LockState::Id)
    {
        ChipLogProgress(Zcl, "Door lock cluster: " ChipLogFormatMEI, ChipLogValueMEI(clusterId));
    }

    if (clusterId == ZCL_ON_OFF_CLUSTER_ID)
    {
        OnOnOffPostAttributeChangeCallback(endpointId, attributeId, value);
    }
}

// add below callback
void OnOnOffPostAttributeChangeCallback(EndpointId endpointId, AttributeId attributeId, uint8_t * value)
{
    VerifyOrExit(attributeId == ZCL_ON_OFF_ATTRIBUTE_ID, ChipLogProgress(Zcl, "Unhandled Attribute ID: '0x%04x", (int)attributeId));
    VerifyOrExit(endpointId == 1 , ChipLogProgress(Zcl, "Unexpected EndPoint ID: `0x%02x'", (int)endpointId));
    if (*value)
    {
        LockMgr().InitiateAction(AppEvent::kEventType_Lock, LockManager::LOCK_ACTION);
    }
    else
    {
        LockMgr().InitiateAction(AppEvent::kEventType_Lock, LockManager::UNLOCK_ACTION);
    }
exit:
    return;
}

```

bzbarsky-apple, 2022-11-08:
>         emberAfWriteAttribute(1, ZCL_ON_OFF_CLUSTER_ID, ZCL_ON_OFF_ATTRIBUTE_ID, (uint8_t *) &newOnOffState, ZCL_BOOLEAN_ATTRIBUTE_TYPE);

May I suggest using the functions in `app-common/zap-generated/attributes/Accessors.h`?  So:

```
   EmberAfStatus OnOffstatus = chip::app:Clusters::OnOff::Attributes::OnOff:Set(newOnOffState);
```

(and `using` as desired to make the namespacing simpler).

>     if (clusterId == ZCL_ON_OFF_CLUSTER_ID)

`if (clusterId == OnOff::Id)`

>     VerifyOrExit(attributeId == ZCL_ON_OFF_ATTRIBUTE_ID, ChipLogProgress(Zcl, "Unhandled Attribute ID: '0x%04x", (int)attributeId));

Again, please use the namespaced ids.  We're trying to remove the ZCL_* bits.

>        LockMgr().InitiateAction(AppEvent::kEventType_Lock, LockManager::LOCK_ACTION);

This seems like a _very_ strange way to change the state of an LED.  This "map the On/Off state tothe lock manager state of the _lock_" is the one weird part I see here.  The rest looks reasonable.

mmt-10, 2022-11-08:
@bzbarsky-apple 
Thank you for your suggestion.
I will try it.
And, ESP32 and other platform examples uses ZCL_ON_OFF_ATTRIBUTE_ID.
I hope examples are update to modern code.

BTW, would you please follow below issue?
[https://github.com/project-chip/connectedhomeip/issues/23414](https://github.com/project-chip/connectedhomeip/issues/23414)

closed 2022-11-10 by mmt-10

bzbarsky-apple, 2022-11-18:
>  I hope examples are update to modern code.

I hope so too, and the ones I reviewed were, but I can't review every PR personally....