project-chip/connectedhomeip #23501: [TC-SU-2.5] Need yaml syntax support for following steps

kowsisoundhar12, 2022-11-04:
### Feature description

```bash
...
[TC-SU-2.5] Need yaml syntax support for following steps

1.Verify that the DUT starts updating its software.
2.Verify that the DUT clears its previously downloaded software image, and resets the UpdateState Attribute to Idle.
```

### Platform

all

### Anything else?

[image]
[image]

kowsisoundhar12, 2022-11-04:
https://github.com/project-chip/connectedhomeip/issues/23249

bzbarsky-apple, 2022-11-10:
How can YAML possibly verify "that the DUT starts updating its software" for an actual device?  @kowsisoundhar12

kowsisoundhar12, 2022-11-16:
@bzbarsky-apple sorry ,let me close this issue ,since we can't verify above steps in yaml

closed 2022-11-16 by kowsisoundhar12