project-chip/connectedhomeip #23506: [TC-SU-3.3] Need yaml syntax support for Following Step

kowsisoundhar12, 2022-11-04:
### Feature description

```bash
...
need an yaml syntax support for following steps:
 1: OTA-R/TH sends a QueryImage Command to the DUT. Protocol supported should only list "BDX". DUT responds with the QueryImageResponse to the OTA-R/TH.
```

### Platform

all

### Anything else?

[image]

kowsisoundhar12, 2022-11-04:
https://github.com/project-chip/connectedhomeip/issues/23249

bzbarsky-apple, 2022-11-10:
I would think this already worsk. In particular, if you send a QueryImage in YAML and the response is not QueryImageResponse, the test will fail right now.

@kowsisoundhar12 what exactly is missing here?

kowsisoundhar12, 2022-11-21:
QueryImage command will work in yaml ,hence closing this issue

closed 2022-11-21 by kowsisoundhar12