project-chip/connectedhomeip #23509: Bridge Action cluster NodeLabel attribute showing UnsupportedWrite response instead Success

manjunath-grl, 2022-11-04:
### Reproduction steps

```bash
Pair DUT and TH
Run test TC-BRBINFO-2.1 againt Bridge-app
in step 18: TH writes "newnode" to NodeLabel attribute, as per spec it is write supported but when executed got UNSUPPORTEDWRITE response instead Success.

Client : Bridge-app
server : Chip-tool
```

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

0c0092309188bcc5718aeb4b0f8f68eb468688ce

### Platform

other

### Anything else?

Diff link: https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:NodeLabel_UnsupportedWrite_Error
Testplan screenshot: 
[image]

Execution logs:
[image]
[exe_log.txt](https://github.com/project-chip/connectedhomeip/files/9941044/exe_log.txt)

bzbarsky-apple, 2022-11-10:
This is writing cluster 0x0039 (Bridged Device Basic Information), attribute id 0x5 (NodeLabel).  That attribute is optional to support for that cluster.

That said, if it _is_ supported it should be writable, and in this case the example app supports it but does not allow writing it.  `emberAfExternalAttributeWriteCallback` returns failure for everything except the on/off attribute.  So this needs to be fixed in the example app.

cjandhyala, 2023-06-26:
@manjunath-grl closing the issue since we executed the test fine in 1.1 , feel free to re-open if you think the issue is still valid.

manjunath-grl, 2023-06-28:
@cjandhyala checked with the latest commit:23a0064eee834ee1e9b0ffc6d243a6a3866c8349 facing the same error: `Test Failure: status value mismatch: expected 0 but got 136(UnsupportedWrite)`
[image]

Can you please reopen this issue again.