project-chip/connectedhomeip #23513: CHIPTool

yangguanming, 2022-11-05:
### Reproduction steps

```bash
Unresolved reference: chip
```

### Bug prevalence

Whenever

### GitHub hash of the SDK that was being used

1

### Platform

android

### Type

Platform Issue

### Anything else?

[image]

jungyee, 2022-11-08:
I have the same problem in my build.

jungyee, 2022-11-08:
I've got the same problem.

matterdish, 2022-11-25:
This error is happenning because the CHIPController.jar and Setuppayloadparser.jar and other .so files is not getting generated. Build using a script given and then change the mattersrcbuild to false. The .jar and .so files will be generated and added to the file.

palak-android, 2023-03-28:
Hello,

I am Building [Android CHIPTool from Android Studio](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/android_building.md#source-files)

I am trying to execute below command from Git Bash terminal and it is getting failed.
`$ TARGET_CPU=arm64 ./scripts/examples/android_app_ide.sh
...
...
+ source scripts/activate.sh
++ setup/bootstrap.sh
scripts/activate.sh: line 1: setup/bootstrap.sh: No such file or directory`

Please let me know what is wrong with command.

Full output:
`Full output:

/Samples/connectedhomeip (master)$ TARGET_CPU=arm64 ./scripts/examples/android_app_ide.sh
+ env
USERDOMAIN=AP
OS=Windows_NT
COMMONPROGRAMFILES=C:\Program Files\Common Files
PROCESSOR_LEVEL=6
PSModulePath=%ProgramFiles%\WindowsPowerShell\Modules;C:\Windows\system32\WindowsPowerShell\v1.0\Modules
CommonProgramW6432=C:\Program Files\Common Files
ANDROID_HOME=/c/Users/test/AppData/Local/Android/Sdk/
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
STTASKFOLDERPATH=Ivanti\PatchSDK\Deployments
LANG=en_US.UTF-8
MSYSTEM_CARCH=x86_64
DISPLAY=needs-to-be-defined
HOSTNAME=7NKJY33
PUBLIC=C:\Users\Public
CONFIG_SITE=/mingw64/etc/config.site
EXEPATH=C:\Users\test\AppData\Local\Programs\Git
UATDATA=C:\Windows\CCM\UATData\D9F8C395-CAB8-491d-B8AC-179A1FE1BE77
MSYSTEM_CHOST=x86_64-w64-mingw32
USERNAME=test
LOGONSERVER=\\INSAMDDC02P
PROCESSOR_ARCHITECTURE=AMD64
LOCALAPPDATA=C:\Users\test\AppData\Local
COMPUTERNAME=7NKJY33
!::=::\
SYSTEMDRIVE=C:
USERPROFILE=C:\Users\test
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
SYSTEMROOT=C:\Windows
USERDOMAIN_ROAMINGPROFILE=AP
OneDriveCommercial=C:\Users\test\OneDrive
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 142 Stepping 12, GenuineIntel
MINGW_PACKAGE_PREFIX=mingw-w64-x86_64
PWD=/c/D/Projects/2023/Matter_Jan_2023/Samples/connectedhomeip
SSH_ASKPASS=/mingw64/libexec/git-core/git-gui--askpass
HOME=/c/Users/test
TMP=/tmp
ANDROID_NDK_HOME=/c/Users/test/AppData/Local/Android/Sdk/ndk/21.4.7075529/
MSYSTEM_PREFIX=/mingw64
OneDrive=C:\Users\test\OneDrive - Arrow Electronics, Inc
ZES_ENABLE_SYSMAN=1
PROCESSOR_REVISION=8e0c
TMPDIR=/tmp
USERDNSDOMAIN=AP.CORP.ARROW.COM
NUMBER_OF_PROCESSORS=8
ProgramW6432=C:\Program Files
COMSPEC=C:\Windows\system32\cmd.exe
APPDATA=C:\Users\test\AppData\Roaming
__PSLockDownPolicy=0
TERM=xterm
SHELL=/usr/bin/bash
TARGET_CPU=arm64
WINDIR=C:\Windows
MINGW_CHOST=x86_64-w64-mingw32
ProgramData=C:\ProgramData
SHLVL=2
PLINK_PROTOCOL=ssh
PROGRAMFILES=C:\Program Files
ACLOCAL_PATH=/mingw64/share/aclocal:/usr/share/aclocal
MANPATH=/mingw64/local/man:/mingw64/share/man:/usr/local/man:/usr/share/man:/usr/man:/share/man
ORIGINAL_TEMP=/tmp
ORIGINAL_TMP=/tmp
ALLUSERSPROFILE=C:\ProgramData
TEMP=/tmp
DriverData=C:\Windows\System32\Drivers\DriverData
MSYSTEM=MINGW64
MINGW_PREFIX=/mingw64
ProgramFiles(x86)=C:\Program Files (x86)
PATH=/c/Users/test/bin:/mingw64/bin:/usr/local/bin:/usr/bin:/bin:/mingw64/bin:/usr/bin:/c/Users/test/bin:/c/Program Files/Common Files/Oracle/Java/javapath:/c/Windows/system32:/c/Windows:/c/Windows/System32/Wbem:/c/Windows/System32/WindowsPowerShell/v1.0:/c/Windows/System32/OpenSSH:/c/Program Files/TortoiseGit/bin:/c/Users/test/AppData/Local/Android/Sdk/platform-tools:/c/Program Files/TortoiseSVN/bin:/c/Program Files/dotnet:/c/Users/test/AppData/Local/Microsoft/WindowsApps:/cmd:/usr/bin/vendor_perl:/usr/bin/core_perl
HOMEDRIVE=C:
PKG_CONFIG_PATH=/mingw64/lib/pkgconfig:/mingw64/share/pkgconfig
INFOPATH=/usr/local/info:/usr/share/info:/usr/info:/share/info
HOMEPATH=\Users\test
ORIGINAL_PATH=/mingw64/bin:/usr/bin:/c/Users/test/bin:/c/Program Files/Common Files/Oracle/Java/javapath:/c/Windows/system32:/c/Windows:/c/Windows/System32/Wbem:/c/Windows/System32/WindowsPowerShell/v1.0:/c/Windows/System32/OpenSSH:/c/Program Files/TortoiseGit/bin:/c/Users/test/AppData/Local/Android/Sdk/platform-tools:/c/Program Files/TortoiseSVN/bin:/c/Program Files/dotnet:/c/Users/test/AppData/Local/Microsoft/WindowsApps:/cmd
_=/usr/bin/env
+ '[' -z /c/Users/test/AppData/Local/Android/Sdk/ ']'
+ '[' -z /c/Users/test/AppData/Local/Android/Sdk/ndk/21.4.7075529/ ']'
+ '[' -z arm64 ']'
+ source scripts/activate.sh
++ setup/bootstrap.sh
scripts/activate.sh: line 1: setup/bootstrap.sh: No such file or directory`

robrecha, 2024-05-03:
I am experiencing the same issue attempting to build Android Chip Tool from Android Studio via Ubuntu in WSL2. I am following this guide: [https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/android_building.md](url). I cannot see any issues until I attempt to make the project from within Android Studio.
[image]
Per comments from @matterdish, it looks like missing files (CHIPController.jar, Setuppayloadparser.jar and other .so files). I followed @matterdish's instructions (changing matterSdkSourceBuild in gradle.properties to true, rather than  mattersrcbuild) but still no luck.
Can anyone suggest where I might have gone wrong? Thx

aniketkumar04, 2024-07-17:
@robrecha Getting same error. how you fixed this.. any information on the same will be helpful Thankyou.