project-chip/connectedhomeip #23524: [1.0] SRP update error

Leo2442926161, 2022-11-07:
### Reproduction steps

```bash
1. build EFR32 firmware with commit 26a054c31144478e3c27391c06b0dc3230bb0882 (HEAD -> v1.0-branch, origin/v1.0-branch)
2. falsh v2.6 TH firmware to raspberry pi.
3. do pairing.
```

### Bug prevalence

almost every time

### GitHub hash of the SDK that was being used

commit 26a054c31144478e3c27391c06b0dc3230bb0882 (HEAD -> v1.0-branch, origin/v1.0-branch)

### Platform

efr32, raspi

### Platform Version(s)

matter 1.0 and TH v2.6

### Type

Core SDK Interopability Issue

### Anything else?

Dear all,

we tried the old commit it works fine, for example, we tried the te9 branch and commit d47c8f7a3ac84f2cc04e5411f064baeb9fe8c93f. both pairing are smoothly, but it fails on commit 26a054c31144478e3c27391c06b0dc3230bb0882 almost everytime.

I have attached the log on chip-tool and efr32 side, please take a look, thank you.

[chiptool log.txt](https://github.com/project-chip/connectedhomeip/files/9951080/chiptool.log.txt)
[EFR32 ThreadNetworkSetup error.log](https://github.com/project-chip/connectedhomeip/files/9951081/EFR32.ThreadNetworkSetup.error.log)

mkardous-silabs, 2022-11-07:
@Leo2442926161 Have you tried commissioning with the chip-tool command `ble-pairing` built from the same hash as the end device?

jmartinez-silabs, 2022-11-07:
[19:33:56.215]ÊÕ¡û¡ô<info  > [DL] SRP Client was started, detected server: fd12:7c3b:cbb8:02ca:0f26:43f1:a4e5:237c

[19:33:56.319]ÊÕ¡û¡ô<info  > [DL] Disconnect Event for handle : 1
<info  > [DL] BLE GATT connection closed (con 1, reason 4104)

[19:33:59.248]ÊÕ¡û¡ô<error > [DL] SRP update error: timed out waiting on server response

This is usually a failure from the OTBR as it fails to respond to the device request for the SRP service to advertise. This sometimes happens when the same Node ID is reused multiple times and the past service failed to be deleted properly. 

Also what ot-rcp are you using? I have seen those kinds of issues also when an old version of ot-rcp is used.

you can give it a try with a standalone OTBR instead of the docker image.

closed 2023-05-18 by stale[bot]