project-chip/connectedhomeip #23525: [Documentation] Setup using google Home

fariouche, 2022-11-07:
### Documentation issues

```bash
I see that google home is now supporting Matter...
I tried using an esp32 example, and google home can detect the matter object over BLE!
However, that is all. It ask for a pin code or a qrcode, and I see nowhere how to set this pin or generate this qrcode :(

Any help is appreciated!
Thanks!
```

### Platform

esp32

### Anything else?

Using chiptool on android works (as it does not ask for the qrcode or pincode)

lboue, 2022-11-07:
Helo @fariouche,

You can check esp32 Serial console to get the qrcode.
Check espressif website: [Developing with the SDK](https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html)

fariouche, 2022-11-07:
wahoo! There is a qrcode in the page and I can get further in the steps! thank you!
Now I have a crash, but most likely because of me...

closed 2022-11-07 by fariouche