project-chip/connectedhomeip #23537: Compilation issue when trying to build lighting-app on nRF board

phonnakasturi-apple, 2022-11-07:
### Reproduction steps / Feature

```bash
With SHA b278e34bbf4720ac6008776fec2aba8a3a75ebd8

Steps to reproduce:

1. cd connectedhomeip
2. git reset ---hard b278e34bbf4720ac6008776fec2aba8a3a75ebd8
3. export PKG_CONFIG_PATH="/opt/homebrew/opt/openssl@3/lib/pkgconfig"
4. source scripts/bootstrap.sh
5. source scripts/activate.sh
6. python3 scripts/setup/nrfconnect/update_ncs.py --update

This throws the following error:

"ERROR: update failed for project cmock"
```

### Platform

darwin

### Type

Manually tested with SDK

### Anything else?

Attaching the complete logs

[noridc_error_logs.txt](https://github.com/project-chip/connectedhomeip/files/9956135/noridc_error_logs.txt)

phonnakasturi-apple, 2022-11-08:
Note: This last worked with SHA 0c0092309188bcc5718aeb4b0f8f68eb468688ce

andy31415, 2022-11-10:
@phonnakasturi-apple could you bisect the build and find out the first failing commit?

phonnakasturi-apple, 2022-11-11:
@andy31415 Following is the first failing commit;

SHA 236e14e65030b8e12d49b6501352972eff42ad60

```
commit 236e14e65030b8e12d49b6501352972eff42ad60
Author: Kamil Kasperczyk <66371704+kkasperczyk-no@users.noreply.github.com>
Date:   Fri Oct 28 00:42:27 2022 +0200

    [nrfconnect] Switched to using nRF Connect SDK 2.1.1 version (#23333)
    
    Changed recommended nRF Connect SDK revision to 2.1.1 and did some
    necessary alignments:
    * nRF platform
    - Added support for SPI NOR external flash
    - Enabled entering QSPI NOR deep sleep mode for nRF52 family
    * nRF examples
    - Removed flash controller configuration from examples DTS
    - Aligned DFUOverSMP callback API and optimized power consumption
    by disabling QSPI NOR
    - Fixed emitting leave event at factory reset in window-app
    - Moved OTARequestor initialization to kDnssdPlatformInitialized
    event handler to avoid race condition (sending NotifyUpdateApplied
    if SRP client is not yet fully initialized)
    - Changed MRP local active retry interval for lighting-app to avoid
    packet stall while switching from Thread child to router
```

Damian-Nordic, 2022-11-29:
This error message seems relevant:
```
Cloning into '/Users/priyankahonnakasturi/chip-nordic-tools/nrfconnect/sdk-nrf/test/cmock/vendor/c_exception'...
Submodule path 'vendor/c_exception': checked out '71b47be7c950f1bf5f7e5303779fa99a16224bb6'
fatal: --stdin can only be used when fetching from one remote
```

kkasperczyk-no, 2022-12-01:
@phonnakasturi-apple in another issue you faced a problem with lighting-app compilation. Does it mean you were able to checkout nRF Connect SDK dependent repos successfully? If yes, could you share what solved the problem?

phonnakasturi-apple, 2022-12-01:
@kkasperczyk-no This issue is present in https://github.com/project-chip/connectedhomeip/issues/23817#issuecomment-1333534351 as well.

By following the steps as mentioned below;

1. cd connectedhomeip
2. git reset ---hard b278e34bbf4720ac6008776fec2aba8a3a75ebd8
3. export PKG_CONFIG_PATH="/opt/homebrew/opt/openssl@3/lib/pkgconfig"
4. source scripts/bootstrap.sh
5. source scripts/activate.sh
6. python3 scripts/setup/nrfconnect/update_ncs.py --update

I get;
```
ERROR: update failed for project cmock
Command '['west', 'update']' returned non-zero exit status 1.
```
By ignoring the error and making sure python version is set to 3.8 and executing the next 2 steps;

8. cd examples/lighting-app/nrfconnect
9. west build -b nrf52840dk_nrf52840 --pristine always

Gives the error mentioned in https://github.com/project-chip/connectedhomeip/issues/23817

kkasperczyk-no, 2022-12-02:
@phonnakasturi-apple, there is few things that would be worth to check:
1. Try to manually update the sdk-nrf repository, not using the `update_ncs.py` script. You can do that in the following way:
    1. Navigate to the `/Users/priyankahonnakasturi/chip-nordic-tools/nrfconnect/sdk-nrf/nrf`.
    2. Make sure with `git status` command that output is `HEAD detached at v2.1.1`.
    3. Invoke `west update` command.
    4. Verify if the failure is the same, as using the script
2. If that fails there is another thing you could verify. As @Damian-Nordic mentioned **--stdin can only be used when fetching from one remote** looks suspicious. Failing `cmock` is git submodule that is by default fetched from the origin remote. The failure log for me looks like there would not be appropriate remote to fetch data.
    1. Go to this failing `cmock` repo location on disk (I guess `Users/priyankahonnakasturi/chip-nordic-tools/nrfconnect/sdk-nrf/test/cmock/vendor/c_exception`) and see the output from commands: `git remote show` and `git status`. I think that it should be `origin` and `HEAD detached at 71b47be`.
    2. If no remote is available I would recommend to add it manually with `git remote add origin https://github.com/throwtheswitch/cexception.git`
    3. Try to do `git fetch origin` and then `git checkout 71b47be`.

I have a hope it is going to work or at least will give us some more information.

phonnakasturi-apple, 2022-12-02:
@kkasperczyk-no When I tried step 1 , I got the same error as I would with update_ncs.py script

With Step 2 and sub step i, I get ->

```
c_exception % git remote show
throwtheswitch
% git status
HEAD detached at refs/heads/manifest-rev
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   ../unity (new commits)

no changes added to commit (use "git add" and/or "git commit -a")
```

cjandhyala, 2022-12-02:
@kkasperczyk-no I have the same versions in my Mac, But I get the same error when I build the lighting app.. my build command is `west build -b nrf52840dk_nrf52840 --pristine always -- -DCONFIG_CHIP_LIB_SHELL=y`

```
nrf % git status                                 
HEAD detached at 71b47be

```
```
nrf % git status
HEAD detached at v2.1.1

```

LuDuda, 2022-12-04:
I believe it might be the `west` issue. I had the same `cmock` error using west 0.12.0. After just upgrading to west 0.14.0 (`pip3 install west==0.14.0`) `west update` succeeds.

Could you try it? If that works, we will update the required version for pigweed.

LuDuda, 2022-12-08:
@phonnakasturi-apple could you please try to upgrade the west tool and see if that helps?

cjandhyala, 2022-12-08:
@LuDuda When we updated the arm GNU compiler to the latest version ([gcc-arm-none-eabi-10.3-2021.10-mac.tar.bz2](https://developer.arm.com/-/media/Files/downloads/gnu-rm/10.3-2021.10/gcc-arm-none-eabi-10.3-2021.10-mac.tar.bz2?rev=58ed196feb7b4ada8288ea521fa87ad5&hash=62C9BE56E5F15D7C2D98F48BFCF2E839D7933597)) , we are able to build the app, is it fine to use the latest tool chain ?