project-chip/connectedhomeip #23541: Why can't you subscribe to more than one property

moon20704, 2022-11-08:
### Reproduction steps

```bash
when we subcribe to more than one property, the second will replace the first,the log is in "Anythings else" item.
```

### Bug prevalence

it appear  inevitably

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip        master branch

### Platform

android

### Anything else?

the error log is that:

D  Found matching exchange: 11177i, Delegate: 0x779dafe710
D  Rxd Ack; Removing MessageCounter:88395640 from Retrans Table on exchange 11177i
D  SubscribeResponse is received
D  SubscribeResponseMessage =
D  {
D  	SubscriptionId = 0x9a5eaf4f,
D  	MaxInterval = 0x3c,
D  	InteractionModelRevision = 1
D  }
D  Subscription established with SubscriptionID = 0x9a5eaf4f MinInterval = 10s MaxInterval = 60s Peer = 01:0000000000000006
D  MoveToState ReadClient[0x779dafe700]: Moving to [Subscripti]
D  Refresh LivenessCheckTime for 61500 milliseconds with SubscriptionId = 0x9a5eaf4f Peer = 01:0000000000000006
D  <<< [E:11177i M:88395641 (Ack:69692302)] (S) Msg TX to 1:0000000000000006 [D1F6] --- Type 0000:10 (SecureChannel:StandaloneAck)
D  (S) Sending msg 88395641 on secure session with LSID: 40128
D  Flushed pending ack for MessageCounter:69692302 on exchange 11177i
I  AnalyticsMessageHandler msg wath = 3
I  delayFlush timeDelayMills = 15000
I  sendData
I  rawMessage = [{"_track_id":88571330,"time":1667878097877,"type":"track","lib":{"$lib":"Android","$lib_version":"2.0.7","$lib_method":"code","$lib_detail":"com.sensorsdata.analytics.android.sdk.AbstractSensorsDataAPI##trackEventInternal##null##291"},"event":"app_eleclick","properties":{"$screen_orientation":"portrait","$os_version":"10","$model":"M2003J15SC","$os":"Android","$screen_width":1080,"$screen_height":2340,"$manufacturer":"Xiaomi","$app_version":"4.7.2","account_id":"39747b0dfd25b156391751bdc5658ddeb47c8ad7ebaeb7ef32a7fd291f3c08dc","iot_deviceid":"c9f15e888751e160cf670ebe6cb74ba7cbbd062abdee23b784b04712aa1aaa66","$wifi":true,"$network_type":"WIFI","$build_version":29,"$project_id":"Yv1_nhtgSxuOFmf-wyXXJw","$auto_id":"16310844829821460ac85-c2de-4dee-9c69-fdd3aa7f236c","$lib_version":"2.0.7","$lib":"Android","$registration_code":"IN","page_id":"H1","element_id":"H1-6"},"_flush_time":1667893525580}]
I  sendHttpRequest path = https://uat.collector.ap-south-1.bd.tcljd.com/v2/secured?project=Yv1_nhtgSxuOFmf-wyXXJw&country=in
I  sendHttpRequest rawMessage = [{"_track_id":88571330,"time":1667878097877,"type":"track","lib":{"$lib":"Android","$lib_version":"2.0.7","$lib_method":"code","$lib_detail":"com.sensorsdata.analytics.android.sdk.AbstractSensorsDataAPI##trackEventInternal##null##291"},"event":"app_eleclick","properties":{"$screen_orientation":"portrait","$os_version":"10","$model":"M2003J15SC","$os":"Android","$screen_width":1080,"$screen_height":2340,"$manufacturer":"Xiaomi","$app_version":"4.7.2","account_id":"39747b0dfd25b156391751bdc5658ddeb47c8ad7ebaeb7ef32a7fd291f3c08dc","iot_deviceid":"c9f15e888751e160cf670ebe6cb74ba7cbbd062abdee23b784b04712aa1aaa66","$wifi":true,"$network_type":"WIFI","$build_version":29,"$project_id":"Yv1_nhtgSxuOFmf-wyXXJw","$auto_id":"16310844829821460ac85-c2de-4dee-9c69-fdd3aa7f236c","$lib_version":"2.0.7","$lib":"Android","$registration_code":"IN","page_id":"H1","element_id":"H1-6"},"_flush_time":1667893525580}]
I  query = crc=668098241&gzip=1&data_list=H4sIAAAAAAAAAH1Ty27bMBD8F7rHWCApiaR8C9oGcYA%2BgByaoigEPlY2G0k0KMpuEOTfu5QTJLn0QonD0ewOd%2FTrkbQpanvfekc2StWSlSW9IMkPQDZMCKmkog2uEsGHA4Jk4ZML0ntDNo%2Fkw%2FIkl6OLAUUuFqA9Qpx8GPGAF7SQL%2FAAaR%2BwErHBwQvoIGnfL%2BBQTDBOIU5OJ13oUfcPydsJ3xb1YnL3xaWZcgvp9sz8hMzL79vVaunr8xHGtB0TRPx2tRrnHlfeMPJ0QSCfYRl9OLTQg%2B394uMQwwFi8jAtdiYbAcY2RI9snc4mDiGivE%2B55TC9ccdohgZ0kw184ZSWN6y%2B%2FXgmvr%2BXZ%2BWTd2mPl0sVfQX34Hd7bI6XVUYHPc4depwjRBS58zoMPmvk3l%2BrV4UsOMLa2jCPaRkiKRtZSUNd53htWC3KhsmaGWdrUSvnwFTSKu0kGA1GQldyLZHbsK60VDmLej4knMrRW1gUbdOxGpRSqANMUNsJScGAsEZWRktrjKOCa%2BMAeGmkqgytJONaM621ELnxk%2B882aQ4A25GSKcQ79vnRP3YXm0zx8y%2Bd6%2F2eIMYTucP2GdrP4%2BsHfdpd%2Ft3%2FnY1dOvTw93dzWm5lzmFM4eJEq%2B2qhRvFGeVoNqqem25g3WF%2Fa0bK5p151yp0TYvhf1%2FZN%2BPMMLO5%2FTlWLRLhjdk%2BzWHSO%2FgXP%2Ba4RbjNcDLQK7ZWuT8tV0%2FT%2Fv2zb%2FVlDWva0Wffv8DaeEX%2F4kDAAA%3D
I  [okhttp]:check permission begin!
W  ClassLoader referenced unknown path: system/framework/mediatek-cta.jar
I  [okhttp] e:java.lang.ClassNotFoundException: com.mediatek.cta.CtaUtils
I  errorMessage = Connection error: java.net.UnknownHostException: Unable to resolve host "uat.collector.ap-south-1.bd.tcljd.com": No address associated with hostname
E  Subscription Liveness timeout with SubscriptionID = 0xd85396dd, Peer = 01:0000000000000006
D  SecureSession[0x77804cda80]: MarkAsDefunct Type:2 LSID:40093
D  SecureSession[0x77804cdb68]: MarkAsDefunct Type:2 LSID:40094
D  SecureSession[0x77804cdc50]: MarkAsDefunct Type:2 LSID:40095
D  SecureSession[0x77804cdd38]: MarkAsDefunct Type:2 LSID:40096
D  SecureSession[0x77804cde20]: MarkAsDefunct Type:2 LSID:40097
D  SecureSession[0x77804cdf08]: MarkAsDefunct Type:2 LSID:40098
D  SecureSession[0x77804cdff0]: MarkAsDefunct Type:2 LSID:40099
D  SecureSession[0x77804ce0d8]: MarkAsDefunct Type:2 LSID:40100
D  SecureSession[0x77804ce1c0]: MarkAsDefunct Type:2 LSID:40101
D  SecureSession[0x77804ce2a8]: MarkAsDefunct Type:2 LSID:40102
D  SecureSession[0x77804ce390]: MarkAsDefunct Type:2 LSID:40103
D  SecureSession[0x77804ce478]: MarkAsDefunct Type:2 LSID:40104
D  SecureSession[0x77804ce560]: MarkAsDefunct Type:2 LSID:40105
D  SecureSession[0x77804ce648]: MarkAsDefunct Type:2 LSID:40106
D  SecureSession[0x77804ce730]: MarkAsDefunct Type:2 LSID:40107
D  SecureSession[0x77804ce818]: MarkAsDefunct Type:2 LSID:40108
D  SecureSession[0x77804ce900]: MarkAsDefunct Type:2 LSID:40109
D  SecureSession[0x77804ce9e8]: MarkAsDefunct Type:2 LSID:40110
D  SecureSession[0x77804cead0]: MarkAsDefunct Type:2 LSID:40111
D  SecureSession[0x77804cebb8]: MarkAsDefunct Type:2 LSID:40112
D  SecureSession[0x77804ceca0]: MarkAsDefunct Type:2 LSID:40113
[... 149 lines cut ...]
I  [okhttp] e:java.lang.ClassNotFoundException: com.mediatek.cta.CtaUtils
I  errorMessage = Connection error: java.net.UnknownHostException: Unable to resolve host "uat.collector.ap-south-1.bd.tcljd.com": No address associated with hostname
I  AnalyticsMessageHandler msg wath = 3
I  delayFlush timeDelayMills = 15000
I  sendData
I  rawMessage = [{"_track_id":88571330,"time":1667878097877,"type":"track","lib":{"$lib":"Android","$lib_version":"2.0.7","$lib_method":"code","$lib_detail":"com.sensorsdata.analytics.android.sdk.AbstractSensorsDataAPI##trackEventInternal##null##291"},"event":"app_eleclick","properties":{"$screen_orientation":"portrait","$os_version":"10","$model":"M2003J15SC","$os":"Android","$screen_width":1080,"$screen_height":2340,"$manufacturer":"Xiaomi","$app_version":"4.7.2","account_id":"39747b0dfd25b156391751bdc5658ddeb47c8ad7ebaeb7ef32a7fd291f3c08dc","iot_deviceid":"c9f15e888751e160cf670ebe6cb74ba7cbbd062abdee23b784b04712aa1aaa66","$wifi":true,"$network_type":"WIFI","$build_version":29,"$project_id":"Yv1_nhtgSxuOFmf-wyXXJw","$auto_id":"16310844829821460ac85-c2de-4dee-9c69-fdd3aa7f236c","$lib_version":"2.0.7","$lib":"Android","$registration_code":"IN","page_id":"H1","element_id":"H1-6"},"_flush_time":1667893555668}]
I  sendHttpRequest path = https://uat.collector.ap-south-1.bd.tcljd.com/v2/secured?project=Yv1_nhtgSxuOFmf-wyXXJw&country=in
I  sendHttpRequest rawMessage = [{"_track_id":88571330,"time":1667878097877,"type":"track","lib":{"$lib":"Android","$lib_version":"2.0.7","$lib_method":"code","$lib_detail":"com.sensorsdata.analytics.android.sdk.AbstractSensorsDataAPI##trackEventInternal##null##291"},"event":"app_eleclick","properties":{"$screen_orientation":"portrait","$os_version":"10","$model":"M2003J15SC","$os":"Android","$screen_width":1080,"$screen_height":2340,"$manufacturer":"Xiaomi","$app_version":"4.7.2","account_id":"39747b0dfd25b156391751bdc5658ddeb47c8ad7ebaeb7ef32a7fd291f3c08dc","iot_deviceid":"c9f15e888751e160cf670ebe6cb74ba7cbbd062abdee23b784b04712aa1aaa66","$wifi":true,"$network_type":"WIFI","$build_version":29,"$project_id":"Yv1_nhtgSxuOFmf-wyXXJw","$auto_id":"16310844829821460ac85-c2de-4dee-9c69-fdd3aa7f236c","$lib_version":"2.0.7","$lib":"Android","$registration_code":"IN","page_id":"H1","element_id":"H1-6"},"_flush_time":1667893555668}]
[... cut ...] LQK7ZWuT8tV0%2FT%2Fv2zb%2FVlLWksqqffv8DTnE%2Fy4kDAAA%3D
I  [okhttp]:check permission begin!
W  ClassLoader referenced unknown path: system/framework/mediatek-cta.jar
I  [okhttp] e:java.lang.ClassNotFoundException: com.mediatek.cta.CtaUtils
I  errorMessage = Connection error: java.net.UnknownHostException: Unable to resolve host "uat.collector.ap-south-1.bd.tcljd.com": No address associated with hostname
E  Subscription Liveness timeout with SubscriptionID = 0x9a5eaf4f, Peer = 01:0000000000000006

bzbarsky-apple, 2022-11-10:
The SDK supports this fine.  Not sure whether the Android bits expose the "KeepSubscriptions" configuration....

yunhanw-google, 2023-02-08:
@joonhaengHeo  could you help take a look at this? Generally speaking, the JAVA/Kotlin API support multiple attributes, but there is UI bug regarding how to describe multiple attributes? maybe add a wildcard mode?

joonhaengHeo, 2023-02-09:
In my case, I add several attributes to the wildcard function below.
Example : 
`listOf(attribute1, attribute2)`

https://github.com/project-chip/connectedhomeip/blob/1ab43dffb01702e748219b22bf43d59e669c7673/src/controller/java/src/chip/devicecontroller/ChipDeviceController.java#L497

However, in this case, only string(object) and tlv format were returned in java and kotlin, so tlv decoding function was additionally implemented internally and used. (Using ZAP)

String(Object)
https://github.com/project-chip/connectedhomeip/blob/1ab43dffb01702e748219b22bf43d59e669c7673/src/controller/java/src/chip/devicecontroller/model/AttributeState.java#L42

TLV :
https://github.com/project-chip/connectedhomeip/blob/1ab43dffb01702e748219b22bf43d59e669c7673/src/controller/java/src/chip/devicecontroller/model/AttributeState.java#L49

joonhaengHeo, 2023-03-15:
I modified in CHIPTool UI.
Please refer below PR.
#25677

closed 2023-03-21 by yunhanw-google