project-chip/connectedhomeip #23555: Using the examples with Google Home

guylevin, 2022-11-09:
### Reproduction steps

```bash
Hello  
  
I've tried some of the examples using ESP32_DevkitC_V4 and Google Home  
  
The board is flashed correctly, and I'm able to pair it afterward to using the Google Home app  
  
I can see in the serial monitor and also in the router that the board is connected to the wifi  
  
I can see the device in the google home app  
  
But when I'm trying to switch on the light (for example in the lighting app) using the google home app, it just says "something went wrong"  
I don't see any logs related to it in the serial console
```

### Bug prevalence

Every time

### GitHub hash of the SDK that was being used

fe9a8ec2f4d383ea244300ffb9953de32e952492

### Platform

esp32

guylevin, 2022-11-09:
please delete

ZH-neutron-star, 2022-11-10:
please, how do u solve this issue ?

guylevin, 2022-11-10:
@ZH-neutron-star I didn't yet

devtn97, 2022-11-11:
I think google home use matter old version, I build new chiptool to control it work.

tcarmelveilleux, 2022-11-12:
Matter support for Google Home ecosystem is still under public preview and the final release is upcoming. Until there is official support in Google Home app for the final Matter 1.0, some usecases have limited explanations or error handling.

Issues with pre-release Google Home cannot be addressed in SDK. Closing issue since it is not related to SDK directly.

closed 2022-11-12 by tcarmelveilleux