project-chip/connectedhomeip #23563: [1.0] darwin-framework-tool BLE pairing doesn't worked on mac osx ventura 13.0

geoffroyletourneur, 2022-11-09:
### Reproduction steps

```bash
1. Build darwin-framework-tool cmd on mac osx ventura 13.0 commit v1.0-branch
2. Launch darwin-framework-tool
3. Run Pairing CMD pairing ble-wifi and tried to pair esp32 with lightning-app.
```

### Bug prevalence

day

### GitHub hash of the SDK that was being used

3897c9b12f4f16a8ad6bf73aa8f8d2272b60124a

### Platform

darwin

### Type

Platform Issue

### Anything else?

2022-11-09 18:49:30.599119+0100 darwin-framework-tool[55430:542772] [all] 🟢 [1668016170599] [55430:542772] CHIP: [IN] Message appended to BLE send queue
2022-11-09 18:49:30.599110+0100 darwin-framework-tool[55430:543852] didWriteValueForCharacteristic: characteristic service:<CBService: 0x600001718a00, isPrimary = YES, UUID = FFF6>
[1668016170599] [55430:542772] CHIP: [IN] Message appended to BLE send queue
2022-11-09 18:49:30.599145+0100 darwin-framework-tool[55430:542772] [all] 🟢 [1668016170599] [55430:542772] CHIP: [SC] Sent PBKDF param request
[1668016170599] [55430:542772] CHIP: [SC] Sent PBKDF param request
2022-11-09 18:49:30.599161+0100 darwin-framework-tool[55430:543852] didWriteValueForCharacteristic: characteristic descriptors:(null)
2022-11-09 18:49:30.599181+0100 darwin-framework-tool[55430:543852] didWriteValueForCharacteristic: characteristic properties:8
2022-11-09 18:49:30.599194+0100 darwin-framework-tool[55430:543852] didWriteValueForCharacteristic: characteristic isNotifying:0
2022-11-09 18:49:30.599204+0100 darwin-framework-tool[55430:543852] didWriteValueForCharacteristic: characteristic isBroadcasted:0
2022-11-09 18:49:30.599225+0100 darwin-framework-tool[55430:543852] [all] 🔴 [1668016170599] [55430:543852] CHIP: [BLE] BLE:Error writing Characteristics in Chip service on the device: [The specified UUID is not allowed for this operation.]
[1668016170599] [55430:543852] CHIP: [BLE] BLE:Error writing Characteristics in Chip service on the device: [The specified UUID is not allowed for this operation.]
2022-11-09 18:49:30.599355+0100 darwin-framework-tool[55430:542772] [all] 🔴 [1668016170599] [55430:542772] CHIP: [IN] Failed to establish BLE connection: ../../../../../../../../../src/platform/Darwin/BleConnectionDelegateImpl.mm:271: Ble Error 0x00000407: GATT write characteristic operation failed
[1668016170599] [55430:542772] CHIP: [IN] Failed to establish BLE connection: ../../../../../../../../../src/platform/Darwin/BleConnectionDelegateImpl.mm:271: Ble Error 0x00000407: GATT write characteristic operation failed

benjaminnestler, 2022-11-10:
Seems to be doubled! See my issue at #23435

bzbarsky-apple, 2022-11-10:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/23435

closed 2022-11-10 by bzbarsky-apple

geoffroyletourneur, 2022-11-10:
@benjaminnestler  @bzbarsky-apple Ok I will see your issue !