project-chip/connectedhomeip #23565: Lighting-App (Master) do not Pair with iOS 16.1

Apollon77, 2022-11-09:
Hi,

While testing we discovered that the Lighting app (compiled o an Ubuntu Machine and run via Ethernet) do not successfully pair with iOS 16.1

In fact yes it could be iOS or it could be Chip ...:-)

When I read the log (see below) correctly then iOS answer with an error 2 to the Case_Sigma2 message.
iOS tells me about an uncertified device that I want to pair with, but then it runs into a timeout .

[chiplog.txt](https://github.com/project-chip/connectedhomeip/files/9975750/chiplog.txt)

PS: Also interesting is that if I choose to use the QR Code then iOS 16.1 tells me that I need a Hub device to connect it. When I use the PIN then the above log happens

Apollon77, 2022-11-10:
BTW: As one info: We experience the same result with a "Node-matter" branch: see https://github.com/mfucci/node-matter/issues/103#issuecomment-1304650359

Apollon77, 2022-11-11:
I enhanced the issue again with informations after a try to pair with an HomePod which was indeed successfull

Apollon77, 2022-11-11:
For completeness here the log of the Homepod (not pairing but reconnection after restart of the lighting app
[out.txt](https://github.com/project-chip/connectedhomeip/files/9992714/out.txt)

carol-apple, 2022-11-12:
@Apollon77 Can you please follow instructions below to file a feedback assistant ticket? 

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

Apollon77, 2022-12-30:
I will recheck this with iOS 16.2 soon.

BTWWhen pairing with a Homepod then it works

Apollon77, 2023-03-10:
Works in the meantime - and in fact is about homepod being there.

closed 2023-03-10 by Apollon77