project-chip/connectedhomeip #23566: Lock-app(v1.0-branch) do not working on Apple Home app

aki-sko, 2022-11-10:
### Reproduction steps

```bash
Hello.
I have Lock-app working on M5Stamp C3.
I'm trying to move the above device to the Home App on my iPhone (iOS 16.1), but it says "uncertified accessory" and I can't control it.
does anyone know the solution?

#Problem
##expected behavior
  Add M5Stamp with Lock-app written to iOS (16.1) Home App and make it controllable.
##Steps to reproduce
  1.Prepare M5Stamp equipped with the following.
  2.Select “Add accessories” in the Home App (HomePod mini is connected) on iPhone (iOS16.1) and read the QR code.
  3."Uncertified device" is displayed when adding an accessory and the device cannot be controlled
```

### Bug prevalence

Whenever do I this

### GitHub hash of the SDK that was being used

3897c9b

### Platform

esp32

Apollon77, 2022-11-10:
so, seems to affect also other apps, not just lighting (see #23565)

carol-apple, 2022-11-12:
@aki-sko Can you please follow instructions below to file a feedback assistant ticket? 

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

closed 2023-05-20 by stale[bot]