project-chip/connectedhomeip #23572: [BUG] SDK is using HKDF based on RFC5869 while the Matter specification specifies HKDF according SP 800-56Cr2 option 2.

wimnxp, 2022-11-10:
### Reproduction steps

```bash
The Matter 1.0 specification specifies Crypto_KDF to be the KDF as specified in Section 4.1 Option 2 of NIST 800-56C. 
All crypto implementations in the SDK seem to be using the HKDF according RFC5869 which would result in a different output compared to using the HKDF SP800-56Cr2 option 2, as the internals of the crypto algorithm are different.
```

### Bug prevalence

Calling Crypto_KDF

### GitHub hash of the SDK that was being used

7b284c027bd5499ca3c5be00d51f83715e8969d6

### Platform

core

### Anything else?

There is also an issue raised on the spec (#5708) for this particular topic.

bzbarsky-apple, 2022-11-10:
@tcarmelveilleux @emargolis @turon @balducci-apple

bzbarsky-apple, 2023-05-12:
Spec was changed.

closed 2023-05-12 by bzbarsky-apple