project-chip/connectedhomeip #23585: ReadEvent command showing build error when executed

manjunath-grl, 2022-11-11:
### Reproduction steps

```bash
Tried adding readEvent command in script TC-ACL-2.5 but facing build error showing that " TypeError: value.split is not a function " attached logs and Diff link please take a look at it and if I am missing anything please let me know.
```

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

008df4d5ad0e16896cdb155cf8d310d8747fb974

### Platform

other

### Anything else?

Diff link:  https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:ReadEvent_Build_Error

YAML screenshot: 
[image]

Execution logs:
[image]
[exe_log.txt](https://github.com/project-chip/connectedhomeip/files/9992680/exe_log.txt)

bzbarsky-apple, 2022-11-16:
@manjunath-grl Please retest on a SHA from after PR #23574 was merged.

bzbarsky-apple, 2022-11-16:
Also, note that the readEvent response YAML there looks wrong: it has an extra list that is likely making it not actually do any checks.  @manjunath-grl

manjunath-grl, 2022-12-07:
@bzbarsky-apple tried with that latest commit and without list but sill facing regen error seeing " ⛔ TypeError: undefined is not a function" when tried to execute test.
Updated diff link: https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:ReadEvent_Build_Error
Execution logs:
[ACL_Regen_Error.zip](https://github.com/project-chip/connectedhomeip/files/10174519/ACL_Regen_Error.zip)

bzbarsky-apple, 2022-12-09:
It looks like what's left here is that readEvent is just not supported in darwin-framework-tool (e.g. the existing TestEvents.yaml test is disabled there)....

bzbarsky-apple, 2022-12-09:
And in fact there is no Darwin framework API to do `readEvent`.  You should just be disabling any tests that `readEvent` in darwin-framework-tool.  See `examples/darwin-framework-tool/templates/tests/ciTests.json`.

manjunath-grl, 2022-12-14:
Closing the issue ACL test are working fine after disabling them in Darwin CiJson file

closed 2022-12-14 by manjunath-grl