project-chip/connectedhomeip #23586: [BUG] Unable to compile iOS chiptool - redefinition of enumerations errors

kean-apple, 2022-11-11:
### Reproduction steps

```bash
1.open iOS chiptool Xcode project (connectedhomeip/src/darwin/Darwin.xcworkspace)
2. Set Xcode team info
3.Try to compile iOS chiptool get various redefinition of enumerations errors and property has a previous declaration errors

Observed this with 008df4d5ad0e16896cdb155cf8d310d8747fb974 & b278e34bbf4720ac6008776fec2aba8a3a75ebd8
```

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

008df4d5ad0e16896cdb155cf8d310d8747fb974

### Platform

darwin

### Anything else?

[image]

kean-apple, 2022-11-15:
Seems to compile on older Xcode versions eg. 13.4.1 on a different Mac

bzbarsky-apple, 2022-11-16:
@kean-apple Where do the error messages claim the "original" definition is?

My guess is that you are somehow managing to include both the system headers and the local headers or something like that....

closed 2023-06-10 by stale[bot]