project-chip/connectedhomeip #23588: [BUG] Lighting-bulb example running on Nordic 52840 does not respond on the iPhone Home App after being added.

marcorussi, 2022-11-11:
### Reproduction steps

```bash
1. Having an Apple HomePod Mini and an iPhone, follow the steps in https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md
2. Build the lighting-bulb example for Nordic nRF5840 DK and flash the device
3. Add the Matter accessory to the iPhone Home App via QR code (open the link printed on UART console from the accessory)
4. The accessory is added but looks unresponsive on the Home App. However the UART console proves that the accessory is still running and LED (bulb) can be controlled by the dedicated button on the DK board.
...
```

### Bug prevalence

Every time I do this

### GitHub hash of the SDK that was being used

aa9457e6b94b735076dff6297176183bf9780177

### Platform

nrf

### Platform Version(s)

nRF5840 DK (PCA10056) v2.0.1

### Anything else?

I followed all the steps and the nrf SDK was updated as indicated by the guide: https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md

After a while I get the following UART log from the board:
[image]

Here is a screenshot of the services. Note that i also have a rasperry pi 4 with another DK as RCP running for other experiments:
[image]

This is what I get from the board:
[image]
[image]
[image]

Let me know if you need more info.
I am not sure if it is a bug or I did something wrong.
Thank you for your help.

marcorussi, 2022-11-16:
@bzbarsky-apple iPhone 14 IOS 16.0.3 and HomePod Mini version 16.1.
Thanks.

nivi-apple, 2022-11-18:
@marcorussi Please use SVE2 sha - c279578c5bc37f117335aa96cec6c5552f070cc0. We currently support this SHA with ios 16.0.3. Please let me know if that doesn't work for you and we will look into it.

marcorussi, 2022-11-19:
Hi @nivi-apple thank you for your help. I have tried the commit you suggested and I still have the same issue. The device does not respond in the Home App. I can successfully add it though as before.

closed 2023-06-10 by stale[bot]