project-chip/connectedhomeip #23596: [BUG] Linux thermostat-app example not recognized in iOS 16.2 beta

lboue, 2022-11-14:
### Reproduction steps

```bash
@vivien-apple @bzbarsky-apple
Will Thermostat type devices be supported soon in iOS 16.2 beta?
During my last test it was not recognized.

Reproduction steps:
1. Build Linux thermostat-app example 
2. Run hermostat-app example 
3. Add Matter accessory in an iPad

`
~/cd connectedhomeip
source scripts/activate.sh 
cd ~/connectedhomeip/examples/thermostat/linux
gn gen out/debug
ninja -C out/debug

sudo ./out/debug/thermostat-app --wifi --discriminator 2803   --trace_log 1 --trace_file thermostat-app.log
````
```

### Bug prevalence

each time

### GitHub hash of the SDK that was being used

030bd45a51587ec464efd8637f76a43886bdaca0

### Platform

other

### Anything else?

- iPad running iOS 16.2 beta 2 version

bzbarsky-apple, 2022-11-16:
>  Will Thermostat type devices be supported soon in iOS 16.2 beta?

We can't comment on future releases; standard Apple policy...

nivi-apple, 2022-11-16:
@lboue Thermostat is already supported via the all-clusters-app which you can configure as a thermostat. As for the linux app, I am going to look into it. will let you know what I find.

beckmx, 2023-04-28:
Hello guys, where exactly is the thermostat example, I used one of the examples and for iOS looks like doesn’t show the temperature control, I don’t know why, it looks like this:
[image]
I would like to test the Linux one, if you can point me what file did you used to see if it only affects my implementation

lboue, 2023-04-29:
@beckmx You can use this directory: [thermostat/linux](https://github.com/project-chip/connectedhomeip/tree/master/examples/thermostat/linux).

closed 2023-04-29 by lboue