project-chip/connectedhomeip #23600: [mrp] MRP default timeouts should be relaxed for Thread devices

turon, 2022-11-15:
### Reproduction steps

```bash
The [MRP timeout defaults](https://github.com/project-chip/connectedhomeip/blob/master/src/messaging/ReliableMessageProtocolConfig.h#L46) determine how quickly a client will retry a dropped message and are meant to be adjusted based on latency expectations of a product.  The defaults currently assume a high availability, low latency device and any device that doesn't fit these properties should choose override the default configuration with a more appropriate custom setting.

Thread devices are a common class in Matter, and require a higher value for their MRP parameters.   Large IPv6 packets will be broken into multiple 6LoWPAN fragments, each of which may travel over multiple mesh hops and need to be reassembled by the recipient. This results in longer latency expectations for Thread devices as show here: https://openthread.io/testing/performance-dashboard/average-latency

Given Thread has a 2000ms reassembly timeout and average round-trip time for a 512 byte message over 3 hops can exceed 260ms, the proposal is to add an `#ifdef THREAD` to the `CHIP_CONFIG_MRP_LOCAL_ACTIVE_RETRY_INTERVAL` and `CHIP_CONFIG_MRP_LOCAL_IDLE_RETRY_INTERVAL` and to set those to a larger value for builds that use a Thread interface.

Setting the MRP parameters too low for Thread devices can cause a client to send a burst of retries, congesting the link, leading to general communication failure.
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

9a41c9c

### Platform

core

Damian-Nordic, 2022-12-08:
@turon I did some analysis and my recommendation would be to increased the SAI/SII attributes on Thread devices by at least 800ms, and additionally increase the MRP backoff time on the Thread sender side depending on the packet size: see https://docs.google.com/document/d/1FKptf1If7sZmTDAQc9DN1unMxX07dCOkWvdnrA9Di14/edit?usp=sharing.

Let me know your opinion on that.

SkipAshton, 2022-12-12:
There are two questions here - 
1.  What should the MRP default be?
2. What should a sleeping device set ifs MRP backoff time to for if condition?

For normal operations setting the default low is generally ok for efficient communications.  However, we know there is limited performance data published for thread network round trip time, and it depends on message size.

For a sleeping device, the larger parameter is meant to be how quickly a device wakes to check for a message when it is expecting one (active interval).  It was common to see this set to 500 ms wake time so a sleeping thread device was counting on its parent to hold the message for up to 500 ms before it asked for it.  If the device knows it is doing this then it should be setting its MRP default to a number larger than 500 ms so messages are not resent while the child has not had time to pick it up.  

The Silabs published round trip time showed ~150 ms out to 7 hops with 300 byte packets.  So having a default of 200 ms for normal devices and letting sleepies set it longer based on their active sleep interval seemed to be ok.  
Given the longer round trip time in this new data - we need to understand why is it so much longer (without even involving the sleepy device yet).
Do we think the Silabs data is out of data and OpenThread round trip time is this much worse?  The OpenThread performance dashboard would seem to show data better than this but it looks a bit aged also.

Damian-Nordic, 2023-01-20:
@SkipAshton I need to find time to get back to my report and do more testing to be able to answer your questions fully, but just for the record:
1. This issue is not about SEDs, SEDs already have the retry intervals increased in the current implementation.
2. Some reasons why 300ms is not enough for Thread devices (in my opinion):
- as you mentioned, Silabs used packets up to 300B, and in Matter we can easily find much bigger packets. E.g. `Sigma2` and `Sigma3` have about 700B if I recall correctly, not to mention BDX which by defaults sends 1024B blocks.
- also, they might not test scenarios in which both the request and response are big which can be the case when e.g. one sends multiple read requests in a single message.
- notice that the standalone ACK timeout is configured to 200ms, so if processing a message takes close to that, there's pretty short window left for delivering the ACK.