project-chip/connectedhomeip #23608: [BUG] Facing PASESession.cpp:254: CHIP Error 0x00000032: Timeout error over Ble-wifi pairing in chip-tool

Survensa, 2022-11-15:
### Reproduction steps

```bash
1. Bring up dut(sample-app) - ./chip-all-clusters-app --wifi
2. Pair chip-tool with DUT - ./chip-tool pairing ble-wifi 1 <SSID> <pwd> 20202021 3841
3. Check the log.
...
```

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

commit 9a41c9c3d971797010ab9de4eb04804015674fb0 (HEAD -> master, origin/master, origin/HEAD)

### Platform

raspi

### Anything else?

Error: `

> CHIP:TOO: Run command failure: ../../third_party/connectedhomeip/src/protocols/secure_channel/PASESession.cpp:254: CHIP Error 0x00000032: Timeout

Attached are logs for reference:
1. [wifi-issue.txt](https://github.com/project-chip/connectedhomeip/files/10012768/wifi-issue.txt)
2. [Wifi issue.txt](https://github.com/project-chip/connectedhomeip/files/10012772/Wifi.issue.txt)

bzbarsky-apple, 2022-11-16:
@Survensa Is this a duplicate of https://github.com/project-chip/connectedhomeip/issues/23438 ?

bzbarsky-apple, 2022-11-16:
Duplicate of #23438