project-chip/connectedhomeip #23716: [BUG] Matter protocol design flaw with privacy and message extensions

Flavien, 2022-11-22:
### Reproduction steps

```bash
I may be missing something, but as far as I can tell, there is a flaw in the Matter specification that will prevent privacy from being used in conjunction with message extensions.

With the P flag (privacy) enabled, the protected payload includes the following fields:

    M = Message Counter || [Source ID] || [Destination ID] || [Message Extensions]

When decrypting a privacy-protected message, the four fields above need to be located. However the `Message Extensions` field is of variable length, and its length is encoded as the first two bytes of the `Message Extensions` field. This means the length is encrypted and it's not possible to determine where it end before decrypting, but it's not possible to decrypt is without knowing where it ends.

That means privacy can't be used in conjunction with message extensions without a non-backwards compatible revision of the Matter message format.

Please tell me if I am missing something.
```

### Bug prevalence

Affects core protocol specification

### GitHub hash of the SDK that was being used

n/a

### Platform

ameba, android, cc13x2_cc26x2, darwin, efr32, esp32, freeRTOS, IMX8, k32w, nrf, python, raspi, vscode, windows, other, core

### Platform Version(s)

1.0

bzbarsky-apple, 2022-11-29:
This is tracked (as a spec issue) in https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/5566

closed 2022-11-29 by bzbarsky-apple

Flavien, 2022-12-01:
Thanks @bzbarsky-apple, any chance I could be granted access to issue so I can follow?

bzbarsky-apple, 2022-12-09:
@Flavien I believe you have to be a CSA member, and probably at the Participant level (see https://csa-iot.org/become-member/) to access that repository...