project-chip/connectedhomeip #23746: [BUG] chip-tool Segmentation fault

lbuque, 2022-11-24:
### Reproduction steps

```bash
1. source scripts/bootstrap.sh
2. source ./scripts/activate.sh
3. cd examples/chip-tool/
4. gn gen out
5. ninja -C out
6. cd out/
7. ./chip-tool pairing ble-wifi 0x7283  GL-MT1300-44e 88888888 20202021 3840

Are my steps to run chip-tool correct, and if not, what should I do?
```

### Bug prevalence

must appear

### GitHub hash of the SDK that was being used

9a41c9c3d971797010ab9de4eb04804015674fb0

### Platform

other

### Platform Version(s)

mint linux 20

### Anything else?

```log
[1669284666.196442][187164:187164] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1669284666.196504][187164:187164] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1669284666.196517][187164:187164] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1669284666.196525][187164:187164] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1669284666.196567][187164:187164] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-0udutD)
[1669284666.196634][187164:187164] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1669284666.196639][187164:187164] CHIP:DL: NVS set: chip-counters/reboot-count = 4 (0x4)
[1669284666.196754][187164:187164] CHIP:DL: Got Ethernet interface: enp2s0
[1669284666.196845][187164:187164] CHIP:DL: Found the primary Ethernet interface:enp2s0
[1669284666.196935][187164:187164] CHIP:DL: Got WiFi interface: wlo1
[1669284666.198344][187164:187164] CHIP:DL: Found the primary WiFi interface:wlo1
[1669284666.198357][187164:187164] CHIP:IN: UDP::Init bind&listen port=0
[1669284666.198367][187164:187164] CHIP:IN: UDP::Init bound to port=39339
[1669284666.198370][187164:187164] CHIP:IN: UDP::Init bind&listen port=0
[1669284666.198376][187164:187164] CHIP:IN: UDP::Init bound to port=60141
[1669284666.198378][187164:187164] CHIP:IN: BLEBase::Init - setting/overriding transport
[1669284666.198380][187164:187164] CHIP:IN: TransportMgr initialized
[1669284666.198385][187164:187164] CHIP:FP: Initializing FabricTable from persistent storage
[1669284666.198405][187164:187164] CHIP:TS: Last Known Good Time: 2022-11-24T18:02:01
[1669284666.198485][187164:187164] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0xCA80C6A7EA0F05FD, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1669284666.198644][187164:187164] CHIP:ZCL: Using ZAP configuration...
[1669284666.199020][187164:187164] CHIP:CTL: System State Initialized...
[1669284666.199029][187164:187164] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1669284666.199038][187164:187164] CHIP:CTL: Setting attestation nonce to random value
[1669284666.199042][187164:187164] CHIP:CTL: Setting CSR nonce to random value
[1669284666.199050][187164:187164] CHIP:IN: UDP::Init bind&listen port=5550
[1669284666.199055][187164:187164] CHIP:IN: UDP::Init bound to port=5550
[1669284666.199057][187164:187164] CHIP:IN: UDP::Init bind&listen port=5550
[1669284666.199062][187164:187164] CHIP:IN: UDP::Init bound to port=5550
[1669284666.199064][187164:187164] CHIP:IN: TransportMgr initialized
[1669284666.199089][187164:187166] CHIP:DL: CHIP task running
[1669284666.199144][187164:187166] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1669284666.199151][187164:187166] CHIP:CTL: Setting attestation nonce to random value
[1669284666.199160][187164:187166] CHIP:CTL: Setting CSR nonce to random value
[1669284666.199657][187164:187166] CHIP:CTL: Generating NOC
[1669284666.199803][187164:187166] CHIP:FP: Validating NOC chain
[1669284666.200088][187164:187166] CHIP:FP: NOC chain validation successful
[... 53 lines cut ...]
[1669284668.202976][187164:187164] CHIP:FP: Forgetting fabric 0x1
[1669284668.202992][187164:187164] CHIP:TS: Pending Last Known Good Time: 2022-11-24T18:02:01
[1669284668.203071][187164:187164] CHIP:TS: Previous Last Known Good Time: 2022-11-24T18:02:01
[1669284668.203079][187164:187164] CHIP:TS: Reverted Last Known Good Time to previous value
[1669284668.203102][187164:187164] CHIP:CTL: Shutting down the commissioner
[1669284668.203109][187164:187164] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1669284668.203128][187164:187164] CHIP:CTL: Shutting down the controller
[1669284668.203136][187164:187164] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1669284668.203223][187164:187164] CHIP:DMG: IM WH moving to [Uninitialized]
[1669284668.203231][187164:187164] CHIP:DMG: IM WH moving to [Uninitialized]
[1669284668.203238][187164:187164] CHIP:DMG: IM WH moving to [Uninitialized]
[1669284668.203244][187164:187164] CHIP:DMG: IM WH moving to [Uninitialized]
[1669284668.203249][187164:187164] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1669284668.203273][187164:187164] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1669284668.203290][187164:187164] CHIP:FP: Shutting down FabricTable
[1669284668.203300][187164:187164] CHIP:TS: Pending Last Known Good Time: 2022-11-24T18:02:01
[1669284668.203342][187164:187164] CHIP:TS: Previous Last Known Good Time: 2022-11-24T18:02:01
[1669284668.203349][187164:187164] CHIP:TS: Reverted Last Known Good Time to previous value
[1669284668.203449][187164:187164] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-fr0QoH)
[1669284668.203644][187164:187164] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1669284668.203664][187164:187164] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1669284668.203674][187164:187164] CHIP:DL: Inet Layer shutdown
[1669284668.203681][187164:187164] CHIP:DL: BLE shutdown
Segmentation fault (core dumped)
```

bzbarsky-apple, 2022-11-29:
It looks like there was no device with discriminator `3840` around, so the BLE discovery failed.

You should try running under a debugger to see why that leads to a crash during shutdown, but presumably we are shutting down while there is BLE stuff still going on or something like that....

sid514-m, 2023-01-04:
I am also facing the same issue but instead of Chip-tool my Lighting app is crashing during the BLE commissioning, I am providing the SSID and PSK from the chip tool command which is thrown on another device, and the LinuxNetworkCommissioningDelegate prints the string of SSID and then crashes on the other device. 

I am using the same commands as given in the README files.

closed 2023-08-11 by stale[bot]