project-chip/connectedhomeip #23747: [Feature] Add ControllerTests to OIS unit-tests

ATmobica, 2022-11-24:
### Feature description

```bash
...
Building ControllerTest fails for OIS unit tests.
The python environment for generating the data model is wrong, not contain the required packages.

FAILED: gen/src/controller/data_model/app/PluginApplicationCallbacks.h gen/src/controller/data_model/app/callback-stub.cpp 
/workspaces/CHIP/.environment/pigweed-venv/bin/python3.9 ../../../../../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../../../../../ --current-path ../../../../../controller/data_model --default-toolchain=//config/openiotsdk/chip-gn/toolchain:openiotsdk --current-toolchain=//config/openiotsdk/chip-gn/toolchain:openiotsdk --capture-output --python-virtualenv-config python/gen/third_party/pigweed/repo/pw_env_setup/pigweed_build_venv/venv_metadata.json --python-dep-list-files gen/src/controller/data_model/data_model_codegen_metadata_path_list.txt -- ../../../../../../scripts/codegen.py --generator cpp-app --output-dir gen/src/controller/data_model --expected-outputs gen/src/controller/data_model/data_model_codegen.expected.outputs ../../../../../controller/data_model/controller-clusters.matter
Traceback (most recent call last):
  File "/workspaces/CHIP/src/test_driver/openiotsdk/unit-tests/build/chip_build/../../../../../../scripts/codegen.py", line 16, in <module>
    import click
ModuleNotFoundError: No module named 'click'
```
There are also issues with TestReadChunking and TestWriteChunking which fail in specific use cases.
For now, the ControllerTest is skipped. We need to investigate and fix all issues.
```

### Platform

other

ATmobica, 2022-11-24:
@pan- 
@andreilitvin Please, assign this issue to me. I'll deal with it soon.

andy31415, 2024-05-08:
This issue still exists.