project-chip/connectedhomeip #23756: [Workflow] Unable to complete bootstrap on M1 mac mini

rosahay-silabs, 2022-11-25:
### Workflow issues

```bash
1. `git clone --recurse-submodules https://github.com/project-chip/connectedhomeip.git`
2. `brew install openssl pkg-config`
3. `export PKG_CONFIG_PATH=$PKG_CONFIG_PATH:"/opt/homebrew/opt/openssl@3/lib/pkgconfig"`
4. `source ./scripts/bootstrap.sh`
```

### Platform

darwin

### Anything else?

```shell
$ uname -a
Darwin mac0017172 21.6.0 Darwin Kernel Version 21.6.0: Mon Aug 22 20:20:05 PDT 2022; root:xnu-8020.140.49~2/RELEASE_ARM64_T8101 arm64
$ python3 --version
Python 3.9.6
```

```log
Building wheels for collected packages: psutil, watchdog
  Building wheel for psutil (pyproject.toml): started
  error: subprocess-exited-with-error
  
  × Building wheel for psutil (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  Building wheel for psutil (pyproject.toml): finished with status 'error'
  ERROR: Failed building wheel for psutil
  Building wheel for watchdog (setup.py): started
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  Building wheel for watchdog (setup.py): finished with status 'error'
  ERROR: Failed building wheel for watchdog
  Running setup.py clean for watchdog
Failed to build psutil watchdog
ERROR: Could not build wheels for psutil, which is required to install pyproject.toml-based projects
['/Users/rosahay/Developer/matter/.environment/pigweed-venv/bin/python', '-m', 'pip', 'install', '--log', '/Users/rosahay/Developer/matter/.environment/pigweed-venv/pip-requirements.log', '--requirement=/Users/rosahay/Developer/matter/scripts/requirements.txt', '--constraint=/Users/rosahay/Developer/matter/scripts/constraints.txt'] {'stdout': <_io.TextIOWrapper name=4 mode='w+' encoding='UTF-8'>, 'stderr': -2}

None
Traceback (most recent call last):
  File "/Users/rosahay/Developer/matter/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 795, in <module>
    sys.exit(main())
  File "/Users/rosahay/Developer/matter/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 787, in main
    return EnvSetup(**vars(parse())).setup()
  File "/Users/rosahay/Developer/matter/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 457, in setup
    result = step(spin)
  File "/Users/rosahay/Developer/matter/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 606, in virtualenv
    if not virtualenv_setup.install(
  File "/Users/rosahay/Developer/matter/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 274, in install
    pip_install('--log', os.path.join(venv_path, 'pip-requirements.log'),
  File "/Users/rosahay/Developer/matter/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 243, in pip_install
    return _check_call(cmd)
  File "/Users/rosahay/Developer/matter/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 88, in _check_call
    subprocess.check_call(args, **kwargs)
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/Users/rosahay/Developer/matter/.environment/pigweed-venv/bin/python', '-m', 'pip', 'install', '--log', '/Users/rosahay/Developer/matter/.environment/pigweed-venv/pip-requirements.log', '--requirement=/Users/rosahay/Developer/matter/scripts/requirements.txt', '--constraint=/Users/rosahay/Developer/matter/scripts/constraints.txt']' returned non-zero exit status 1.
```

Attaching logs for reference
[bootstrap.log](https://github.com/project-chip/connectedhomeip/files/10089517/bootstrap.log)

rosahay-silabs, 2022-11-29:
@woody-apple @bzbarsky-apple , anyone I can reach out to in order to get this resolved ?

bzbarsky-apple, 2022-11-29:
@rosahay-silabs Is this fallout from https://github.com/project-chip/connectedhomeip/pull/23279 by any chance?  Can you check whether this worked before that PR?

rosahay-silabs, 2022-12-02:
> @rosahay-silabs Is this fallout from #23279 by any chance? Can you check whether this worked before that PR?

I currently do not have access to the mac mini. I will revert by mid next week.

rosahay-silabs, 2022-12-05:
> @rosahay-silabs Is this fallout from #23279 by any chance? Can you check whether this worked before that PR?

Verified with `2a588c72b904f16c5a114c907bf8b5b91e743fec` and it is still failing on the same step.

bzbarsky-apple, 2022-12-09:
@rosahay-silabs I don't know.  The problem is somewhere in the pigweed python bits...

@rgoliver might know something, but I don't know whether this ping will reach him; Slack might be better if you have that option.

rosahay-silabs, 2022-12-09:
Root cause identified; full installation was XCODE was needed as the `Python.h` header file is being picked up from xcode rather than `/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/Headers/Python.h`

rosahay-silabs, 2022-12-09:
Root cause identified; full installation was XCODE was needed as the Python.h header file is being picked up from xcode rather than `/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/Headers/Python.h`

closed 2022-12-09 by rosahay-silabs