project-chip/connectedhomeip #23767: [Documentation] Links in /REVIEWERS.md Broken

LawrenceQuarshie, 2022-11-26:
### Documentation issues

```bash
1. The reviewers groups in the /REVIEWERS.md file are broken. Clicking on each one of them leads to a 404 page.
```

### Anything else?

https://github.com/orgs/project-chip/teams/reviewers-apple

LawrenceQuarshie, 2022-12-15:
Hello @bzbarsky-apple 

Please links are still broken.

Is there anything that can be done?

Thank you.

bzbarsky-apple, 2022-12-17:
@LawrenceQuarshie The links are not broken.  You're getting 404 because that's what github does if you load a github page that you don't have permissions to view.

https://docs.github.com/en/organizations/organizing-members-into-teams/about-teams#team-visibility says:

> People who are not members of the organization cannot view any teams.

In practice, if you create a PR, it will get reviewed; a number of people look for incoming PRs (though this time of year there's a lot of vacation going on, of course).

closed 2022-12-17 by bzbarsky-apple

LawrenceQuarshie, 2022-12-17:
Thank you very much @bzbarsky-apple 

This is my first contribution to open source so pardon if things didn't go like expected.

I am learning and I am going to get better with time.

bzbarsky-apple, 2022-12-18:
@LawrenceQuarshie No need to apologize!  The github behavior for showing 404 for "permission denied" (which is basically so people can't check whether things they don't have access to exist by getting a different error) is _very_ non-obvious.

LawrenceQuarshie, 2022-12-21:
That is well noted @bzbarsky-apple .