project-chip/connectedhomeip #23776: I encountered a problem. I cross compiled chip lighting app running on 32-bit Linux system. The size is about 1.4M, but this is the size of the executable program. The dynamic library still needs about 8M. The size of the statically compiled chip lighting app is about 7.3M. Is this size normal? I have run the strip command.[1.0]

RedPoppies, 2022-11-28:
### Reproduction steps

```bash
1.Cross compile dynamically linked or statically linked executable programs
```

### Bug prevalence

whenever i do this

### GitHub hash of the SDK that was being used

0b17bce8005fb155e78e243f31111cdaf2c92797

### Platform

other

### Type

Platform Issue

bzbarsky-apple, 2022-11-29:
That size does not sound unexpected, esp. if compiling a debug build.

closed 2022-11-29 by bzbarsky-apple

RedPoppies, 2022-12-05:
@bzbarsky-apple 
Thank you for your reply.
Our flash space is limited. What can we do to reduce the size of the program.
Please give me some suggestions.

bzbarsky-apple, 2022-12-09:
@RedPoppies Wait, was the _target_ of the compilation a "32-bit Linux system"?  Or was that the host?  What is the target here?

RedPoppies, 2022-12-09:
@bzbarsky-apple 
The compiled executable program runs on a 32-bit Linux system (this is an embedded development board).
The compilation host is Ubuntu 20.04 64 bit system.

bzbarsky-apple, 2022-12-17:
I think that if you are trying to run on a space-limited Linux you may need to do some optimization work.  There are all sorts of space-demanding features that are enabled right now if the compilation target is "linux", because the assumption is that generally these are not particularly constrained devices.