project-chip/connectedhomeip #23817: Issue in compilation of lighting-app

phonnakasturi-apple, 2022-11-30:
### Reproduction steps

```bash
With SHA 49654486db9ae006f5c9e2fc89b59c622ad614dd 

While compiling lighting-app, I am running into the following issue;

Error Log;

FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /Users/priyankahonnakasturi/connectedhomeip/examples/lighting-app/nrfconnect/build/mcuboot/zephyr/zephyr_pre0.map 
: && ccache /Users/priyankahonnakasturi/chip-nordic-tools/gnuarmemb/bin/arm-none-eabi-gcc   zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/Users/priyankahonnakasturi/connectedhomeip/examples/lighting-app/nrfconnect/build/mcuboot/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/soc/arm/nordic_nrf/nrf52/libsoc__arm__nordic_nrf__nrf52.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  modules/nrf/lib/fprotect/lib..__nrf__lib__fprotect.a  modules/nrf/drivers/hw_cc310/lib..__nrf__drivers__hw_cc310.a  modules/mcuboot/boot/bootutil/zephyr/libmcuboot_util.a  modules/hal_nordic/nrfx/libmodules__hal_nordic__nrfx.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/Users/priyankahonnakasturi/chip-nordic-tools/gnuarmemb/bin/../lib/gcc/arm-none-eabi/9.2.1/thumb/v7e-m+fp/hard"  -L/Users/priyankahonnakasturi/connectedhomeip/examples/lighting-app/nrfconnect/build/mcuboot/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfpu=fpv4-sp-d16  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  /Users/priyankahonnakasturi/chip-nordic-tools/nrfconnect/sdk-nrf/nrfxlib/crypto/nrf_cc310_platform/lib/cortex-m4/hard-float/no-interrupts/libnrf_cc310_platform_0.9.15.a  /Users/priyankahonnakasturi/chip-nordic-tools/nrfconnect/sdk-nrf/nrfxlib/crypto/nrf_cc310_bl/lib/cortex-m4/hard-float/no-interrupts/libnrf_cc310_bl_0.9.12.a && cd /Users/priyankahonnakasturi/connectedhomeip/examples/lighting-app/nrfconnect/build/mcuboot/zephyr && /Users/priyankahonnakasturi/connectedhomeip/.environment/pigweed-venv/lib/python3.8/site-packages/cmake/data/bin/cmake -E echo
/Users/priyankahonnakasturi/chip-nordic-tools/gnuarmemb/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.bfd: zephyr/zephyr_pre0.elf section `datas' will not fit in region `FLASH'
/Users/priyankahonnakasturi/chip-nordic-tools/gnuarmemb/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.bfd: region `FLASH' overflowed by 64 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
[90/537] Generating gen/app-codegen/cpp-app/app/PluginApplicationCallbacks.h, gen/app-codegen/cpp-app/app/callback-stub.cpp
2022-11-29 16:14:02 INFO    Parsing idl from /Users/priyankahonnakasturi/connectedhomeip/examples/lighting-app/nrfconnect/../lighting-common/lighting-app.matter
2022-11-29 16:14:02 INFO    Running code generator cpp-app
2022-11-29 16:14:02 INFO    File to be generated: app/PluginApplicationCallbacks.h
2022-11-29 16:14:02 INFO    Writing new data to: /Users/priyankahonnakasturi/connectedhomeip/examples/lighting-app/nrfconnect/build/gen/app-codegen/cpp-app/app/PluginApplicationCallbacks.h
2022-11-29 16:14:02 INFO    File to be generated: app/callback-stub.cpp
2022-11-29 16:14:02 INFO    Writing new data to: /Users/priyankahonnakasturi/connectedhomeip/examples/lighting-app/nrfconnect/build/gen/app-codegen/cpp-app/app/callback-stub.cpp
2022-11-29 16:14:02 INFO    Done
[326/537] Building C object modules/openthread/platform/CMakeFiles/openthread_platform.dir/alarm.c.obj
FAILED: modules/mcuboot/mcuboot_subimage-prefix/src/mcuboot_subimage-stamp/mcuboot_subimage-build mcuboot/zephyr/zephyr.hex mcuboot/zephyr/zephyr.elf /Users/priyankahonnakasturi/connectedhomeip/examples/lighting-app/nrfconnect/build/modules/mcuboot/mcuboot_subimage-prefix/src/mcuboot_subimage-stamp/mcuboot_subimage-build /Users/priyankahonnakasturi/connectedhomeip/examples/lighting-app/nrfconnect/build/mcuboot/zephyr/zephyr.hex /Users/priyankahonnakasturi/connectedhomeip/examples/lighting-app/nrfconnect/build/mcuboot/zephyr/zephyr.elf 
cd /Users/priyankahonnakasturi/connectedhomeip/examples/lighting-app/nrfconnect/build/mcuboot && /Users/priyankahonnakasturi/connectedhomeip/.environment/pigweed-venv/lib/python3.8/site-packages/cmake/data/bin/cmake --build . --
[338/537] Linking C static library modules/nrf/drivers/hw_cc310/lib..__nrf__drivers__hw_cc310.a
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /Users/priyankahonnakasturi/connectedhomeip/.environment/pigweed-venv/bin/cmake --build /Users/priyankahonnakasturi/connectedhomeip/examples/lighting-app/nrfconnect/build
priyankahonnakasturi@priyankas-mbp nrfconnect % 

Steps to reproduce:

1. cd ~/connectedhomeip
2. git reset --hard 49654486db9ae006f5c9e2fc89b59c622ad614dd
3. export PKG_CONFIG_PATH="/opt/homebrew/opt/openssl@3/lib/pkgconfig"
4. source scripts/activate.sh
5. source ~/032321_install_nordic_tools.sh
6. python3 scripts/setup/nrfconnect/update_ncs.py --update
7. cd examples/lighting-app/nrfconnect
8. west build -b nrf52840dk_nrf52840 --pristine always
```

### Bug prevalence

Today

### GitHub hash of the SDK that was being used

49654486db9ae006f5c9e2fc89b59c622ad614dd

### Platform

nrf, python

### Platform Version(s)

Darwin

bzbarsky-apple, 2022-11-30:
@phonnakasturi-apple does this run into the same issue for you?
```
scripts/examples/nrfconnect_example.sh lighting-app nrf52840dk_nrf52840
```

It looks like your linked binary ended up too big to fit on the chip you are building for, which is a little surprising given that we do build pretty much this configuration (but with RPC additionally enabled) in CI.

kkasperczyk-no, 2022-11-30:
I'm able to build this target locally on Linux (I don't have mac to verify it for darwin). Generally the region that was overflown should be occupied in almost 100% (99,33% in my case). I assume something may be configured wrong in the local environment, so e.g. different toolchain version than recommended is used and target is linked in slightly different way. @phonnakasturi-apple could you attach the full log from the build and not only the failure part? I will try to compare some details with my own configuration.

phonnakasturi-apple, 2022-11-30:
@bzbarsky-apple With scripts/examples/nrfconnect_example.sh lighting-app nrf52840dk_nrf52840 ; I ran into the same issue.

Attaching the complete log ;
[nordic_issue.txt](https://github.com/project-chip/connectedhomeip/files/10126392/nordic_issue.txt)

phonnakasturi-apple, 2022-11-30:
@kkasperczyk-no Here is the complete log;

[nordic_issue1.txt](https://github.com/project-chip/connectedhomeip/files/10126487/nordic_issue1.txt)

cjandhyala, 2022-11-30:
@bzbarsky-apple @kkasperczyk-no I see the same issue with lighting app while all-cluster-app builds fine with the same SHA using the above script.

kkasperczyk-no, 2022-12-01:
@cjandhyala that's understandable, because by default all-clusters-app is built with OTA Software Update disabled and lighting-app with OTA support enabled. The problem you see is related to the fact of overflowing the region reserved for bootloader that is used for the software update purpose, so that's why it won't be reproduced for all-clusters-app.

Damian-Nordic, 2022-12-01:
Hi, looks like the flash overflow issue occurs on gcc 9.2.1 (and perhaps other 9 minor revisions). We'll try to optimize the flash usage in the future so that the build success doesn't depend so much on the used toolchain, but, anyway, newer versions of Zephyr (and hence nRF Connect SDK) will require using Zephyr-SDK toolchain rather than `arm-none-eabi-gcc` I recommend that you update your scripts to use that toolchain as described in: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/gs_installing.html#install-a-toolchain.

Btw, CI uses Pigweed-supplied GCC 10.3.1 and currently, it seems to be also sufficient.

cjandhyala, 2022-12-06:
we are running 'python3 scripts/setup/nrfconnect/update_ncs.py --update ' and we have the latest zephyr SDK 

=== updating zephyr (zephyr):
HEAD is now at 8689324605 [nrf fromtree] drivers: spi: fix sample syntax

BTW updating the arm GNU compiler to latest , we are able to build the lighting app. Is there any issue in using the latest compiler or do you recommend some version of it ?

PR #24038 (2022-12-13): [nrfconnect] Disable GPIO for MCUBoot
Reduce the size of bootloader by disable GPIO on all boards except nRF7002 DK. The latter needs GPIO because it uses SPI flash.

By the way, remove broken links in READMEs.

Fixes #23817
Fixes #23794
closed 2022-12-13 by andy31415