project-chip/connectedhomeip #23834: Missing cluster error in simulated test

manjunath-grl, 2022-11-30:
### Reproduction steps / Feature

```bash
Seeing cluster missing error when tried to run simulated tests.
1 . Add FLABEL-3.1 simulated test
2 . Run regen command ( ./scripts/tools/zap_regen_all.py )
3 . Shows cluster missing error

Missing cluster:
Account Login
Application Launcher
Fixed Label
Audio Output
Channel
General Commissioning
Network Commissioning
Thread Network Diagnostics
Diagnostic Logs
Door Lock
Low Power
Unit Localization
Time Format Localization
```

### Platform

other

### Platform Version(s)

N/A

### Type

YAML tested

### Anything else?

diff link:
https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:Simulated_Cluster_MissingError
Execution logs:
[image]
[exe_log.txt](https://github.com/project-chip/connectedhomeip/files/10126179/exe_log.txt)

bzbarsky-apple, 2022-12-01:
@raju-apple @manjunath-grl Does it help if you enable the relevant cluster(s) in the placeholder app .zap files?

manjunath-grl, 2022-12-07:
@bzbarsky-apple sure i will try it.

closed 2022-12-13 by andy31415