project-chip/connectedhomeip #23847: [genio] correct dnssd api usage

j8xixo12, 2022-12-01:
### Reproduction steps

```bash
1. Commission with multiple controller
2. Dnssd publish service can not publish service due to out of socket
```

### Platform

other

### Type

Manually tested with SDK

pakls, 2022-12-02:
@bzbarsky-apple This addresses test case failures we found and is necessary.

pakls, 2022-12-02:
@j8xixo12 Could the description be updated specify platform as Genio or MT793x? Currently it's 'other' in the description above. Thanks.

pakls, 2022-12-02:
#23848

j8xixo12, 2022-12-02:
@pakls I can not find Genio or MT793x in the Platform list, so.... I choice other

PR #23848 (2022-12-06): [genio] Correct dnssd api usage
 `Fixes #23847`
This PR contains following changes:
1. Correct dnssd api usage
2. remove unused variables
closed 2022-12-06 by yufengwangca