project-chip/connectedhomeip #23871: TimeoutError: [Errno 60] Operation timed out

chenguojiao, 2022-12-02:
run TARGET_CPU=arm64 ./scripts/examples/android_app_ide.sh 

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...Traceback (most recent call last):
  File "/Users/chenguojiao/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 795, in <module>
    sys.exit(main())
             ^^^^^^
  File "/Users/chenguojiao/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 787, in main
    return EnvSetup(**vars(parse())).setup()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/chenguojiao/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 457, in setup
    result = step(spin)
             ^^^^^^^^^^
  File "/Users/chenguojiao/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 546, in cipd
    cipd_client = cipd_wrapper.init(
                  ^^^^^^^^^^^^^^^^^^
  File "/Users/chenguojiao/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 357, in init
    bootstrap(client, silent, rosetta=rosetta)
  File "/Users/chenguojiao/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 311, in bootstrap
    tmp.write(client_bytes(rosetta))
              ^^^^^^^^^^^^^^^^^^^^^
  File "/Users/chenguojiao/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 240, in client_bytes
    conn.request('GET', path)
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/http/client.py", line 1282, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/http/client.py", line 1328, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/http/client.py", line 1277, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/http/client.py", line 1037, in _send_output
    self.send(msg)
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/http/client.py", line 975, in send
    self.connect()
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/http/client.py", line 1447, in connect
    super().connect()
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/http/client.py", line 941, in connect
    self.sock = self._create_connection(
                ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/socket.py", line 850, in create_connection
    raise exceptions[0]
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/socket.py", line 835, in create_connection
    sock.connect(sa)
TimeoutError: [Errno 60] Operation timed out

platform：MacOs

closed 2022-12-06 by chenguojiao

dujiaoking, 2023-01-10:
一样的出错，如何解决？

Wink-XZ-Li, 2024-06-25:
请问是怎么解决的