project-chip/connectedhomeip #23878: [Platform] iOS

LucaFagan, 2022-12-02:
### Reproduction steps / Feature

```bash
App Store Review Guidelines

Added 2.5.17: “Apps that support Matter must use Apple’s support framework for Matter to initiate pairing. In addition, if you choose to use any Matter software component in your app other than the Matter SDK provided by Apple, the software component must be certified by the Connectivity Standards Alliance for the platform it runs on.”

Can we use the iOS chip tool framework in production app? Is it certified by the Connectivity Standards Alliance?
```

### Platform

darwin

### Platform Version(s)

1.0

### Type

Platform validated

bzbarsky-apple, 2022-12-03:
@LucaFagan I'm not sure what you mean by "iOS chip tool framework".

The Matter.framework that ships with the operating system on iOS/macOS/etc is certified by the CSA and you can use it in a production app.

LucaFagan, 2022-12-03:
@bzbarsky-apple with "iOS chip tool framework" i mean the Matter framework at this path 
https://github.com/project-chip/connectedhomeip/tree/master/src/darwin/Framework

It is used by the iOS demo app to commission devices and interact with them.

bzbarsky-apple, 2022-12-09:
@LucaFagan The Matter.framework shipped with iOS, is compiled built from the source you link to and is certified by the CSA; if you use the system `Matter.framework` in your application you will be using that certified component.

If you take the source and compile it yourself, I don't know offhand what the certification status is.

woody-apple, 2023-04-04:
Thanks @LucaFagan ,

For these types of requests please reach out directly via feedback assistant. This repository is just for the CSA Matter Development and questions.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

closed 2023-04-04 by woody-apple