project-chip/connectedhomeip #23919: [Documentation] Profile installation failed

CryptoGL, 2022-12-06:
### Documentation issues

```bash
1.iPhone shows the profile is no longer valid
2.iOS 16.1.2
3.
...
```

### Anything else?

[image]

[image]

bzbarsky-apple, 2022-12-06:
@CryptoGL I don't believe the profile you are trying to install is needed on iOS 16.1.

CryptoGL, 2022-12-07:
> @CryptoGL I don't believe the profile you are trying to install is needed on iOS 16.1.

[image]
I used the profile downloaded from above.

sidhoo2, 2022-12-08:
@CryptoGL there is no need to install the profile on iOS 16.1. but i  am not success 😭

bzbarsky-apple, 2022-12-08:
@CryptoGL Those instructions are outdated (note they are referencing 15.6).  See https://github.com/project-chip/connectedhomeip/pull/23979 for the updated instructions that would be relevant on 16.1.

closed 2023-06-10 by bzbarsky-apple